AimRT/AimRT #2: Compilation Error

tqolf, 2024-09-25:
## System Information
OS: Ubuntu 22.04, Kernel: 6.8.0-45-generic, Arch: x86_64

1. sys/acl.h not found
- ubuntu
``` bash
sudo apt install libacl1-dev
python3 -m pip install pylibacl
```

AimRT, 2024-09-25:
Can you clarify your issue more clearly? Have you encountered a compilation error? If it's a compilation error preventing you from compiling, can you provide the compilation command and relevant logs?

tqolf, 2024-09-25:
> Can you clarify your issue more clearly? Have you encountered a compilation error? If it's a compilation error preventing you from compiling, can you provide the compilation command and relevant logs?

the command is `./build.sh`, but it failed with `sys/acl.h not found`, but it works now after I installed libacl1-dev.

AimRT, 2024-09-25:
> > Can you clarify your issue more clearly? Have you encountered a compilation error? If it's a compilation error preventing you from compiling, can you provide the compilation command and relevant logs?
> 
> The command is `./build.sh`, but it failed with `sys/acl.h not found`, and it works now after I installed libacl1-dev.

Thanks for the reminder, these are the dependencies required for the iceoryx plugin. Perhaps we might consider summarizing all dependencies in the [cpp build guide](https://docs.aimrt.org/tutorials/quick_start/installation_cpp.html).

closed 2024-09-25 by AimRT