AimRT/AimRT #3: 不玩虚的: 就是中文

Heyangq, 2024-09-25:
关于CMake版本和C++版本
以及依赖其他开源项目的版本可以出说明文档吗？
想尝个鲜，感谢

Heyangq, 2024-09-25:
后续是否支持DDS的插件支持?

AimRT, 2024-09-25:
> 关于CMake版本和C++版本 以及依赖其他库的版本可以出说明文档吗？ 想尝个鲜，感谢

最低支持的 CMAKE 版本是 3.24，大部分的依赖都采用源码进行依赖，直接运行 `./build.sh` 即可从 CMake 文件中配置好的下载地址进行下载，已知本地需要的一些依赖（可能不完全）：

1. 如果要编译 ROS2 相关接口/插件，AimRT 会通过find_package的方式在本地寻找 rclcpp 等依赖，请确保本地安装有[ROS2 Humble](https://docs.ros.org/en/humble/)。
2. 如果要构建 Python 接口、cli 工具等，AimRT 会通过find_package的方式在本地寻找 Python 依赖，请确保本地安装有 Python3及相关的 python 包。
3. zenoh 插件需要本地安装有 rust。
4. boost 依赖由于其内容较大，CMake 会首先检查本地是否有 1.82.0 及以上版本的 boost，如果有则使用本地的，如果没有则进行下载。

如果不需要使用相关的插件和内容，直接在 `build.sh` 中关闭相应的编译选项即可，更多相关内容可以在[文档](https://docs.aimrt.org/tutorials/index.html)中找到。

如果你的网络环境在下载依赖方面存在问题，可以使用我们提供的 url_cn.bashrc 文件将下载地址替换为 gitee 源，如下：

```
source url_cn.bashrc
./build.sh $AIMRT_DOWNLOAD_FLAGS
```

Heyangq, 2024-09-25:
> > 关于CMake版本和C++版本 以及依赖其他库的版本可以出说明文档吗？ 想尝个鲜，感谢
> 
> 最低支持的 CMAKE 版本是 3.24，C++ 编译器需要支持 C++20，经过测试的最低版本是 gcc 11.4，clang 15.0.7，理论上高于此版本的编译器都可以使用。
> 
> 大部分的依赖都采用源码进行依赖，直接运行 `./build.sh` 即可从 CMake 文件中配置好的下载地址进行下载，已知本地需要的一些依赖（可能不完全）：
> 
> 1. 如果要编译 ROS2 相关接口/插件，AimRT 会通过find_package的方式在本地寻找 rclcpp 等依赖，请确保本地安装有[ROS2 Humble](https://docs.ros.org/en/humble/)。
> 2. 如果要构建 Python 接口、cli 工具等，AimRT 会通过find_package的方式在本地寻找 Python 依赖，请确保本地安装有 Python3及相关的 python 包。
> 3. zenoh 插件需要本地安装有 rust。
> 4. boost 依赖由于其内容较大，CMake 会首先检查本地是否有 1.82.0 及以上版本的 boost，如果有则使用本地的，如果没有则进行下载。
> 
> 如果不需要使用相关的插件和内容，直接在 `build.sh` 中关闭相应的编译选项即可，更多相关内容可以在[文档](https://docs.aimrt.org/tutorials/index.html)中找到。
> 
> 如果你的网络环境在下载依赖方面存在问题，可以使用我们提供的 `url_cn.bashrc` 文件将下载地址替换为 gitee 源，如下：
> 
> ```
> source url_cn.bashrc
> ./build.sh $AIMRT_DOWNLOAD_FLAGS
> ```

有没有推荐的Ubuntu版本？Ubuntu22 或者Ubuntu24？

AimRT, 2024-09-25:
> 后续是否支持DDS的插件支持?

当前已有 ros2 插件，支持 channel 和 rpc 后端，分别对应于 ros2 的 topic 和 service，ros2 底层即基于 DDS 进行通信。

短时间内没有原生 DDS 插件支持，但是也会考虑使用情况、用户要求以及开发周期等实际情况进行评估是否进行支持。

AimRT, 2024-09-25:
> 有没有推荐的Ubuntu版本？Ubuntu22 或者Ubuntu24？

Ubuntu 22.04 LTS 是经过测试验证的平台。

closed 2024-09-25 by AimRT