AimRT/AimRT #4: 执行make -j报错

shengqihailuo1, 2024-09-26:
参考： https://docs.aimrt.org/tutorials/quick_start/helloworld_cpp.html 
版本： v0.8.1
执行 make -j 报错如下：
[ 99%] Linking CXX executable aimrt_main
/usr/bin/ld: ../core/libaimrt_runtime_core.a(aimrt_core.cc.o): undefined reference to symbol 'dlclose@@GLIBC_2.2.5'
/usr/bin/ld: /lib/x86_64-linux-gnu/libdl.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [_deps/aimrt-build/src/runtime/main/CMakeFiles/aimrt_runtime_main.dir/build.make:109：_deps/aimrt-build/src/runtime/main/aimrt_main] 错误 1
make[1]: *** [CMakeFiles/Makefile2:2358：_deps/aimrt-build/src/runtime/main/CMakeFiles/aimrt_runtime_main.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
[ 99%] Linking CXX executable helloworld_app
/usr/bin/ld: ../../../_deps/aimrt-build/src/runtime/core/libaimrt_runtime_core.a(aimrt_core.cc.o): undefined reference to symbol 'dlclose@@GLIBC_2.2.5'
/usr/bin/ld: /lib/x86_64-linux-gnu/libdl.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [src/app/helloworld_app/CMakeFiles/helloworld_app.dir/build.make:109：src/app/helloworld_app/helloworld_app] 错误 1
make[1]: *** [CMakeFiles/Makefile2:2505：src/app/helloworld_app/CMakeFiles/helloworld_app.dir/all] 错误 2
[100%] Linking CXX static library libaimrt_protocols_sensor_pb_gencode.a
[100%] Built target aimrt_protocols_sensor_pb_gencode
make: *** [Makefile:156：all] 错误 2

AimRT, 2024-09-26:
你好，可以提供一下你的编译器、操作系统、CMake等工具的版本吗？

可以提供一份你的示例程序代码以便进行复现吗？创建一个 github 仓库或者程序的压缩包都可以。

shengqihailuo1, 2024-09-26:
编译器g++ 11.4.0。操作系统：ubuntu20.04。cmake version 3.28.6。能够支持 c++20。
示例程序代码就是按照这里操作的：https://docs.aimrt.org/tutorials/quick_start/helloworld_cpp.html

AimRT, 2024-09-26:
> 编译器g++ 11.4.0。操作系统：ubuntu20.04。cmake version 3.28.6。能够支持 c++20。 示例程序代码就是按照这里操作的：https://docs.aimrt.org/tutorials/quick_start/helloworld_cpp.html

无法复现

shengqihailuo1, 2024-09-26:
[Uploading ArmRT_demo.zip…]()

AimRT, 2024-09-26:
> Uploading ArmRT_demo.zip…

打不开

shengqihailuo1, 2024-09-26:
> > Uploading ArmRT_demo.zip…
> 
> 打不开

https://github.com/shengqihailuo1/AimRT_demo

AimRT, 2024-09-26:
在 ubuntu 22.04 gcc 11.4 cmake 3.28.3 环境下未能复现该问题
程序正常编译可运行

wtudio, 2024-09-26:
看报错信息，是没有链接 dl，在一些环境上会出现此问题。临时解决方案：

在 https://github.com/shengqihailuo1/AimRT_demo/blob/main/cmake/GetAimRT.cmake#L11 这一行之后，添加以下代码：
```
  target_link_libraries(aimrt_runtime_core PUBLIC ${CMAKE_DL_LIBS})
```

请尝试临时解决方案。

AimRT 将在下一个 fix 版本修复此问题，感谢您的反馈！

AimRT, 2024-09-30:
Fixed in [v0.8.2](https://github.com/AimRT/AimRT/releases/tag/v0.8.2)

closed 2024-09-30 by AimRT