AimRT/AimRT #7: [Bug] 在Windows下例程的名字太长

zmoth, 2024-09-27:
**Version of AimRT**
v0.8.1

**Environments**
MSVC

**Describe the bug**
[build] d:\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppBuild.targets(382,5): error MSB3491: Could not write lines to file "aimrt_examples_cpp_protobuf_channel_benchmark_publisher_module.dir\Debug\aimrt_ex.A3269777.tlog\aimrt_examples_cpp_protobuf_channel_benchmark_publisher_module.lastbuildstate". Path: aimrt_examples_cpp_protobuf_channel_benchmark_publisher_module.dir\Debug\aimrt_ex.A3269777.tlog\aimrt_examples_cpp_protobuf_channel_benchmark_publisher_module.lastbuildstate exceeds the OS max path limit. The fully qualified file name must be less than 260 characters.  [D:\proj\AimRT\build\src\examples\cpp\protobuf_channel\module\benchmark_publisher_module\aimrt_examples_cpp_protobuf_channel_benchmark_publisher_module.vcxproj]
[build] d:\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppBuild.targets(382,5): error MSB3491: Could not write lines to file "aimrt_examples_cpp_protobuf_channel_benchmark_subscriber_module.dir\Debug\aimrt_ex.72F39012.tlog\aimrt_examples_cpp_protobuf_channel_benchmark_subscriber_module.lastbuildstate". Path: aimrt_examples_cpp_protobuf_channel_benchmark_subscriber_module.dir\Debug\aimrt_ex.72F39012.

[build] d:\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppBuild.targets(382,5): error MSB3491: Could not write lines to file "aimrt_examples_cpp_protobuf_rpc_normal_rpc_async_server_module.dir\Debug\aimrt_ex.0BA2E9AB.tlog\aimrt_examples_cpp_protobuf_rpc_normal_rpc_async_server_module.lastbuildstate". Path: aimrt_examples_cpp_protobuf_rpc_normal_rpc_async_server_module.dir\Debug\aimrt_ex.0BA2E9AB.tlog\aimrt_examples_cpp_protobuf_rpc_normal_rpc_async_server_module.lastbuildstate exceeds the OS max path limit. The fully qualified file name must be less than 260 characters.  [D:\proj\AimRT\build\src\examples\cpp\protobuf_rpc\module\normal_rpc_async_server_module\aimrt_examples_cpp_protobuf_rpc_normal_rpc_async_server_module.vcxproj]
[build] d:\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppBuild.targets(382,5): error MSB3491: Could not write lines to file "aimrt_examples_cpp_protobuf_rpc_normal_rpc_async_client_module.dir\Debug\aimrt_ex.DE22842B.tlog\aimrt_examples_cpp_protobuf_rpc_normal_rpc_async_client_module.lastbuildstate". Path: aimrt_examples_cpp_protobuf_rpc_normal_rpc_async_client_module.dir\Debug\aimrt_ex.DE22842B.tlog\aimrt_examples_cpp_protobuf_rpc_normal_rpc_async_client_module.lastbuildstate exceeds the OS max path limit. The fully qualified file name must be less than 260 characters.  

[build] d:\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppBuild.targets(382,5): error MSB3491: Could not write lines to file "aimrt_examples_cpp_protobuf_rpc_normal_rpc_future_client_module.dir\Debug\aimrt_ex.8DC1C9B8.tlog\aimrt_examples_cpp_protobuf_rpc_normal_rpc_future_client_module.lastbuildstate". Path: aimrt_examples_cpp_protobuf_rpc_normal_rpc_future_client_module.dir\Debug\aimrt_ex.8DC1C9B8.tlog\aimrt_examples_cpp_protobuf_rpc_normal_rpc_future_client_module.lastbuildstate exceeds the OS max path limit. The fully qualified file name must be less than 260 characters.  [D:\proj\AimRT\build\src\examples\cpp\protobuf_rpc\module\normal_rpc_future_client_module\aimrt_examples_cpp_protobuf_rpc_normal_rpc_future_client_module.vcxproj]

**To Reproduce**
Steps to reproduce the behavior:
1. 打开AIMRT_BUILD_EXAMPLES
2. 执行编译

zhangyi1357, 2024-09-27:
Windows 下路径长度有限制，可以参考以下流程将路径长度限制解除：

1. 打开注册表编辑器：
  - 按 Win + R 打开运行对话框。
  - 输入 regedit 并按回车键。
2. 导航到注册表项：
  - 在注册表编辑器中，导航到以下路径：
    `HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\FileSystem`
3. 修改 LongPathsEnabled：
  - 在右侧窗格中，找到 LongPathsEnabled 项。如果该项不存在，可以手动创建它。
  - 右键点击右侧窗格，选择 新建 -> DWORD (32-bit) 值。
  - 将新值命名为 LongPathsEnabled。
  - 双击 LongPathsEnabled，将其值设置为 1。
4. 重启计算机：
  - 修改注册表后，需要重启计算机以使更改生效。

zmoth, 2024-09-27:
你好，按照你的解释修改后是能够解决问题的。但是如果能缩短用例的名称长度来解决是不是会更好一些？

closed 2024-09-27 by zmoth

wtudio, 2024-09-27:
感谢你的反馈，我们将考虑在下一个版本缩短用例名称长度