AimRT/AimRT #8: ./build.sh执行失败，cmake配置阶段报错

horance-liu, 2024-09-27:
## OS: Ubuntu 22.04

## Cmake version: 下载cmake官网最新版本

$ cmake --version
cmake version 3.30.3

CMake suite maintained and supported by Kitware (kitware.com/cmake).

## 构建失败：

CMake Error at CMakeLists.txt:179 (find_package):
  By not providing "Findrclcpp.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "rclcpp", but
  CMake did not find one.

  Could not find a package configuration file provided by "rclcpp" with any
  of the following names:

    rclcppConfig.cmake
    rclcpp-config.cmake

  Add the installation prefix of "rclcpp" to CMAKE_PREFIX_PATH or set
  "rclcpp_DIR" to a directory containing one of the above files.  If "rclcpp"
  provides a separate development package or SDK, be sure it has been
  installed.

AimRT, 2024-09-27:
aimrt 默认打开 `AIMRT_BUILD_WITH_ROS2` 和 `AIMRT_BUILD_ROS2_PLUGIN` 选项，如果开启了这两个选项，则需要本地安装有 ros2 humble。如果本地没有安装 ros2 humble，也不想编译 ros2 相关代码，可以关闭这两个选项。

详情请参考 ：https://docs.aimrt.org/tutorials/quick_start/installation_cpp.html

closed 2024-09-27 by AimRT

horance-liu, 2024-09-27:
关闭 AIMRT_BUILD_WITH_ROS2 和 AIMRT_BUILD_ROS2_PLUGIN 选项后

```
$ ./build.sh
CMake Error at /media/tools/cmake-3.30.3-linux-x86_64/share/cmake-3.30/Modules/FindPackageHandleStandardArgs.cmake:233 (message):
  Could NOT find Python3 (missing: Python3_INCLUDE_DIRS Development.Module)
  (found version "3.10.12")
Call Stack (most recent call first):
  /media/tools/cmake-3.30.3-linux-x86_64/share/cmake-3.30/Modules/FindPackageHandleStandardArgs.cmake:603 (_FPHSA_FAILURE_MESSAGE)
  /media/tools/cmake-3.30.3-linux-x86_64/share/cmake-3.30/Modules/FindPython/Support.cmake:4017 (find_package_handle_standard_args)
  /media/tools/cmake-3.30.3-linux-x86_64/share/cmake-3.30/Modules/FindPython3.cmake:601 (include)
  CMakeLists.txt:190 (find_package)
```

zhangyi1357, 2024-09-27:
> 关闭 AIMRT_BUILD_WITH_ROS2 和 AIMRT_BUILD_ROS2_PLUGIN 选项后
>
> ```
> $ ./build.sh
> CMake Error at /media/tools/cmake-3.30.3-linux-x86_64/share/cmake-3.30/Modules/FindPackageHandleStandardArgs.cmake:233 [...]

应该是 python 开发库 python3-dev 未安装，Ubuntu 系统可以使用以下命令安装

```bash
sudo apt-get install python3-dev
```