AimRT/AimRT #9: [Bug] build.sh编译的时候报错

JailbreakFox, 2024-09-27:
**Version of AimRT**
eg. v0.8.1

**Environments**
ubuntu20.04

**Describe the bug**
主要是example下面的ROS相关的示例依赖找不到
```sh
CMake Error at src/tools/ros2_py_gen_aimrt_cpp_rpc/Ros2AimRTRpcGenCode.cmake:34 (target_link_libraries):
  Target "aimrt_protocols_example_ros2_aimrt_rpc_gencode" links to:

    example_ros2::example_ros2__rosidl_generator_cpp
```

```sh
CMake Error at src/plugins/ros2_plugin/CMakeLists.txt:33 (target_link_libraries):
  Target "aimrt_plugins_ros2_plugin" links to:

    ros2_plugin_proto::ros2_plugin_proto__rosidl_generator_cpp
```

JailbreakFox, 2024-09-27:
一定要使用Humble吗？我本地安装的是ROS2-foxy-desktop版本

JailbreakFox, 2024-09-27:
本地安装的是ROS2-foxy-desktop版本

zhangyi1357, 2024-09-27:
> 本地安装的是ROS2-foxy-desktop版本

根据[文档](https://docs.aimrt.org/tutorials/quick_start/installation_cpp.html)，目前仅支持 ros2 humble 版本

closed 2024-09-27 by JailbreakFox