AimRT/AimRT #12: 有视频流传输相关的插件或者功能吗

wuhanhui, 2024-09-30:
希望能将采集的视频，以尽可能小的延迟的方式发送另外一个机器上。这个目前支持吗

peng-zhihui, 2024-10-01:
未来会考虑支持RTSP协议的

wuhanhui, 2024-10-02:
目前我接触到的一些支持RTSP或者webrtc的框架例如[SRS](https://ossrs.io/lts/zh-cn/)，因为服务端都存在缓存，播放一般都会有1秒的延迟，这种比较适合那种直播。对于机器人中实时性要求较高的，目前我都是UDP（采用boost或者poco建个server）直接发送视频流（h264编码），不知道AimRT有什么更好的办法或者思路，非常感谢。

wtudio, 2024-10-08:
这种高实时性场景，当前建议直接使用net_plugin里的udp后端，通过channel发送经过业务编码后的图片帧。如果是本机通信，还可以换成iceoryx后端。各后端性能测试见链接：https://docs.aimrt.org/tutorials/misc/perfomance.html

RTSP、webrtc等跨机视频推流方案，其在设计时就是面向能接受一定延迟的直播场景，并不适合用于实时性要求高的场景。

从更上层的角度来看，由于网络等硬件的限制，不应该在跨机场景对图像视频等大尺寸数据期望较高的实时性。建议从架构层面进行更合理的设计。

AimRT, 2024-10-09:
> > 这种高实时性场景，当前建议直接使用net_plugin里的udp后端，通过channel发送经过业务编码后的图片帧。如果是本机通信，还可以换成iceoryx后端。各后端性能测试见链接：https://docs.aimrt.org/tutorials/misc/perfomance.html
> > RTSP、webrtc等跨机视频推流方案，其在设计时就是面向能接受一定延迟的直播场景，并不适合用于实时性要求高的场景。
> > 从更上层的角度来看，由于网络等硬件的限制，不应该在跨机场景对图像视频等大尺寸数据期望较高的实时性。建议从架构层面进行更合理的设计。
> 
> 性能测试ROS2使用的RMW是什么？

Fastdds

closed 2024-10-09 by AimRT