AimRT/AimRT #18: 数据回环实现

shengqihailuo1, 2024-10-12:
您好，我需要做一个数据回环的功能。要将车上的摄像头/点云/imu数据切块上传到服务器，还要将服务器处理的结果回传到车上。不要求实时性，可以用AimRT框架实现吗？有详细操作示例吗？

zhangyi1357, 2024-10-12:
可以的，实质上就是一个 rpc 请求，使用 AimRT 框架的 rpc 功能即可。

可以使用 net 插件的 http rpc 后端，详细示例可以参考 https://github.com/AimRT/AimRT/tree/main/src/examples/plugins/net_plugin 。

closed 2024-10-12 by AimRT

shengqihailuo1, 2024-10-20:
我的功能需求：将客户端的视频数据回传到服务器，服务器返回处理结果。
基于src/example/cpp/protobuf_rpc/normal_rpc_co_client_module和normal_rpc_co_server_module进行修改。使用的是流式 RPC，对应的.proto文件为：
syntax = "proto3";
package my_package_name;
message SensorDataReq {
    bytes video_chunk_data = 1;  // 视频切块数据
    int32 chunk_index = 2;       // 切块的序号，服务端用来拼接视频
    int32 total_chunks = 3;      // 视频切块总数
    int32 frame_rate = 4;        // 帧率
    string format = 5;           // 视频格式
}
message SensorDataRsp {
    string msg = 1;
    bool result_bool = 2;
}
service my_service {
    rpc my_service_function (stream SensorDataReq) returns (SensorDataRsp);
}

参考：https://docs.aimrt.org/tutorials/interface_cpp/rpc.html，用aimrt封装的cmake生成C++服务代码my_image_armrt_rpc.pb.h :
class my_serviceCoService : public aimrt::rpc::CoServiceBase {
 public:
  my_serviceCoService();
  ~my_serviceCoService() override = default;
  virtual aimrt::co::Task<aimrt::rpc::Status> my_service_function(
      aimrt::rpc::ContextRef ctx_ref,
      const ::my_package_name::SensorDataReq& req,
      ::my_package_name::SensorDataRsp& rsp) {
    co_return aimrt::rpc::Status(AIMRT_RPC_STATUS_SVR_NOT_IMPLEMENTED);
  }
};

继承该类，实现my_serviceImpl函数：
class my_serviceImpl : public my_package_name::my_serviceCoService {
 public:
  my_serviceImpl() = default;
  ~my_serviceImpl() override = default;
  aimrt::co::Task<aimrt::rpc::Status> my_service_function(
      aimrt::rpc::ContextRef ctx,
      const ::my_package_name::SensorDataReq& req,
      //grpc::ServerReader<::my_package_name::SensorDataReq>* reader,
      ::my_package_name::SensorDataRsp& rsp) override;
};
}

问题：aimrt封装的cmake生成C++服务代码my_image_armrt_rpc.pb.h 没有流式 RPC的接口，所以无法传输视频，只能每次传递单张图像吗？

我的代码链接：https://github.com/shengqihailuo1/data_loop

另外，传输上下文（aimrt::rpc::ContextRef ctx）的作用是？在数据闭环的链路中，可以实现什么功能？

shengqihailuo1, 2024-10-20:
上面用的是net 插件的 http rpc 后端。
然后grpc插件（  https://docs.aimrt.org/tutorials/plugins/grpc_plugin.html# ）官网里面说“当前 gRPC 插件仅支持一元 RPC 调用，不支持流式 RPC 调用。“  所以，目前也无法用gRPC 插件传输视频?

zhangyi1357, 2024-10-21:
> 上面用的是net 插件的 http rpc 后端。 然后grpc插件（ https://docs.aimrt.org/tutorials/plugins/grpc_plugin.html# ）官网里面说“当前 gRPC 插件仅支持一元 RPC 调用，不支持流式 RPC 调用。“ 所以，目前也无法用gRPC 插件传输视频?

是的，当前 AIMRT 框架的上层接口并不支持流式 RPC，只有基础的一发一收的 “一元” RPC 接口。

不过我觉得可以想一下是否真的需要流式 RPC，流式 RPC 的主要优点是

1. 不必等待客户端所有内容传输完毕，服务端即可开始进行响应（实时性）；
2. 通过 stream 复用连接，减少 overhead。

针对 2，如果使用 grpc 插件，grpc 调用本身基于 http2，就有连接复用的效果。
针对 1，则需要思考一下相应需求是否需要该特性，我所认知的大部分情况服务端都得接收到完整的消息才能开始响应。

有不同的意见也欢迎探讨！

zhangyi1357, 2024-10-21:
> 问题：aimrt封装的cmake生成C++服务代码my_image_armrt_rpc.pb.h 没有流式 RPC的接口，所以无法传输视频，只能每次传递单张图像吗？

可以使用 repeated 字段在一个 rpc 请求中传递多个画面帧。
 
> 另外，传输上下文（aimrt::rpc::ContextRef ctx）的作用是？在数据闭环的链路中，可以实现什么功能？

ctx 一方面可以设定超时时间（Timeout）、目标请求地址（ToAddr）、序列化格式（SerializationType）等直接与该 RPC 密切相关的信息；另一方面可以包含用户自定义的键值对，用于传递用户希望传递的自定义信息。

Aaron1Zhang, 2024-11-01:
> 您好，我需要做一个数据回环的功能。要将车上的摄像头/点云/imu数据切块上传到服务器，还要将服务器处理的结果回传到车上。不要求实时性，可以用AimRT框架实现吗？有详细操作示例吗？

感觉使用websocket/http更好一些，rpc感觉不太适合大数据传输