AimRT/AimRT #41: make -j 报错 libnghttp2.a

shengqihailuo1, 2024-10-20:
版本:0.8.3 
GetAimRT.cmake:
include(FetchContent)
FetchContent_Declare(
  aimrt
  GIT_REPOSITORY https://github.com/AimRT/aimrt.git
  GIT_TAG v0.8.3)
FetchContent_GetProperties(aimrt)
if(NOT aimrt_POPULATED)
  FetchContent_MakeAvailable(aimrt)
  target_link_libraries(aimrt_runtime_core PUBLIC ${CMAKE_DL_LIBS})
endif()

开启 AIMRT_BUILD_EXAMPLES、AIMRT_BUILD_WITH_PROTOBUF、AIMRT_BUILD_GRPC_PLUGIN 选项编译 AimRT，执行 make -j 报错：
/usr/bin/ld: ../../../../nghttp2-build/lib/libnghttp2.a(nghttp2_session.c.o): relocation R_X86_64_PC32 against symbol `nghttp2_enable_strict_preface' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: 最后的链结失败: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [_deps/aimrt-build/src/plugins/grpc_plugin/CMakeFiles/aimrt_plugins_grpc_plugin.dir/build.make:237：_deps/aimrt-build/src/plugins/grpc_plugin/libaimrt_grpc_plugin.so] 错误 1
make[1]: *** [CMakeFiles/Makefile2:3993：_deps/aimrt-build/src/plugins/grpc_plugin/CMakeFiles/aimrt_plugins_grpc_plugin.dir/all] 错误 2
make: *** [Makefile:156：all] 错误 2

shengqihailuo1, 2024-10-20:
解决方法：
在 GetNghttp2.cmake 中添加：
  set(CMAKE_POSITION_INDEPENDENT_CODE ON)

zhangyi1357, 2024-10-21:
我尝试复现此问题失败了，可以麻烦您提供一份可以复现该错误的代码吗

zhangyi1357, 2024-10-21:
> 解决方法： 在 GetNghttp2.cmake 中添加： set(CMAKE_POSITION_INDEPENDENT_CODE ON)

直接添加此标志可能会将其扩散到其他的 target 中，或许使用 `set_target_properties( PROPERTIES POSITION_INDEPENDENT_CODE ON)` 会更好一点

zhangyi1357, 2024-10-21:
感谢您的反馈，在 https://github.com/AimRT/AimRT/pull/43 中将 PIC 选项直接全局开启，虽然可能会带来微乎其微的性能损失，但是 CMAKE_POSITION_INDEPENDENT_CODE 选项只是一个默认值，如果引用 AimRT 的下游用户仍希望其部分代码的编译不带 PIC 选项，可以自行进行覆盖或者单独指定 `set_target_properties(tgt PROPERTIES POSITION_INDEPENDENT_CODE ON)` 。

closed 2024-10-21 by AimRT