AimRT/AimRT #47: 编译AIMRT_BUILD_ROS2_PLUGIN报错

shengqihailuo1, 2024-10-23:
版本：0.8.1
ubuntu20.04系统,ros2 foxy版本

参考：https://github.com/AimRT/AimRT/tree/main/src/examples/plugins/ros2_plugin
开启 AIMRT_BUILD_EXAMPLES、AIMRT_BUILD_ROS2_PLUGIN 选项编译 AimRT，
执行：./start_examples_plugins_ros2_plugin_protobuf_rpc_server.sh
报错：
AimRT run with exception and exit. Load dynamic lib failed, lib path ./libaimrt_ros2_plugin.so, error info ./libaimrt_ros2_plugin.so: cannot open shared object file: No such file or directory

然后，开启 AIMRT_BUILD_WITH_ROS2 选项编译 AimRT，
执行： cmake ..
报错：
Only the library will be built. To build other components (such as applications and examples), set ENABLE_LIB_ONLY=OFF.
-- Found rosidl_default_generators: 1.0.1 (/opt/ros/foxy/share/rosidl_default_generators/cmake)
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Found rosidl_adapter_proto: 0.0.1 (/usr/local/share/rosidl_adapter_proto/cmake)
CMake Error at /usr/local/share/rosidl_adapter_proto/cmake/rosidl_adapt_proto_interfaces.cmake:65 (message):
  Proto file generation failed, error code: Traceback (most recent call
  last):
    File "/usr/local/lib/rosidl_adapter_proto/rosidl_adapter_proto", line 27, in <module>
      from rosidl_adapter_proto import generate_proto
  ModuleNotFoundError: No module named 'rosidl_adapter_proto'
Call Stack (most recent call first):
  /usr/local/share/rosidl_adapter_proto/cmake/rosidl_adapter_proto-extras.cmake:31 (include)
  /usr/local/share/rosidl_adapter_proto/cmake/rosidl_adapter_protoConfig.cmake:41 (include)
  /usr/local/share/rosidl_typesupport_protobuf_c/cmake/rosidl_typesupport_protobuf_c_generate_interfaces.cmake:21 (find_package)
  /opt/ros/foxy/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  /opt/ros/foxy/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  build/_deps/aimrt-src/src/protocols/example_ros2/CMakeLists.txt:27 (rosidl_generate_interfaces)
-- Configuring incomplete, errors occurred!

zhangyi1357, 2024-10-23:
AimRT 当前针对 ROS2 仅支持 humble 版本。

AimRT, 2024-10-23:
支持 ros2 foxy 版本已经在计划之中。

closed 2024-10-23 by AimRT

shengqihailuo1, 2024-10-23:
> @shengqihailuo1 AimRT 当前针对 ROS2 仅支持 humble 版本。

我安装了docker，ubuntu22.04，ros2 humble版本，编译的时候打开 AIMRT_BUILD_EXAMPLES 、AIMRT_BUILD_WITH_ROS2 和 AIMRT_BUILD_ROS2_PLUGIN选项。
在build目录下执行： ./start_examples_plugins_ros2_plugin_protobuf_rpc_server.sh
报错： source install/share/ros2_plugin_proto/local_setup.bash 找不到文件
解决方法：注释掉这行。
运行：export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/my_ArmRT/build/lib
再执行 ./start_examples_plugins_ros2_plugin_protobuf_rpc_server.sh 就可以了。

zhangyi1357, 2024-10-24:
> 我安装了docker，ubuntu22.04，ros2 humble版本，编译的时候打开 AIMRT_BUILD_EXAMPLES 、AIMRT_BUILD_WITH_ROS2 和 AIMRT_BUILD_ROS2_PLUGIN选项。 在build目录下执行： ./start_examples_plugins_ros2_plugin_protobuf_rpc_server.sh 报错： [...]

应该是没有 make install 导致的，这个 setup 脚本需要 install 之后才有。