AimRT/AimRT #50: About ros2-humble

acuteyaya, 2024-10-24:
When I constructed my project using GetAimRT.cmake, ros2 couldn't find the message type for the back-end node of ros2_plugin, and native ros2 couldn't subscribe to topics using the same type.and install/share is not found in the build folder
[image]

zhangyi1357, 2024-10-24:
The `install/share` directory is likely missing because the `make install` step wasn't executed.

针对自定义的消息类型，需要运行相应消息类型的 setup 脚本，例如https://github.com/AimRT/AimRT/blob/cb01a34047c52a45708fc872c5fa40adad575dd9/src/examples/plugins/ros2_plugin/install/linux/bin/start_examples_plugins_ros2_plugin_ros2_chn_pub.sh#L5

shengqihailuo1, 2024-10-24:
I have a camera driver that publishes a native ros2 topic, can I subscribe to this topic and forward camera data to a remote server via rpc? Is it modified with the case of "native ros2 client calls aimrt server"?
https://github.com/AimRT/AimRT/tree/main/src/examples/plugins/ros2_plugin

zhangyi1357, 2024-10-24:
> I have a camera driver that publishes a native ros2 topic, can I subscribe to this topic and forward camera data to a remote server via rpc? Is it modified with the case of "native ros2 client calls aimrt server"?https://github.com/AimRT/AimRT/tree/main/src/examples/plugins/ros2_plugin

Q1: Yes
Q2: This is also feasible, but note that the AimRT service name has a specific format {proto_type}:/{ServiceName}/{FunctionName}, and will be rewritten to some extent (due to some special characters). Therefore, you need to run `ros2 service list` to see the correct service name.

closed 2024-10-24 by AimRT