AimRT/AimRT #53: [Fea] libstdc++.so.6链接错误：version `GLIBCXX_3.4.32' not found，是否考虑更新依赖安装文档

DongPF1992, 2024-10-25:
操作系统：ubuntu22.04.05LTS
cmake版本：3.30.5（手动安装）
gcc版本：11.4.0

运行报错：
`AimRT run with exception and exit. Load dynamic lib failed, lib path ./libpkg1.so, error info /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.32' not found (required by xxx用户目录xxx/agibot_x1_infer/build/install/lib/libxyber_controller.so)`

参考解决方案：
https://stackoverflow.com/questions/76974555/glibcxx-3-4-32-not-found-error-at-runtime-gcc-13-2-0，即
```
sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt-get update
sudo apt-get install --only-upgrade libstdc++6
```

zhangyi1357, 2024-10-25:
感谢您的反馈！

这个是因为使用的二进制文件使用 gcc-13 编译生成的，所以必须得安装新版本的 glibc 和 libstdc++，不属于 AimRT 的依赖，AimRT 支持 gcc 11.4 及以上的版本。

wtudio, 2024-10-25:
此问题是因为 libxyber_controller.so 这个二进制文件由 gcc-13 编译，所以要求本地需要有gcc13环境。此文件是业务模块带上来的，不属于aimrt框架的依赖。

DongPF1992, 2024-10-25:
> 感谢您的反馈！
> 
> 这个是因为使用的二进制文件使用 gcc-13 编译生成的，所以必须得安装新版本的 glibc 和 libstdc++，不属于 AimRT 的依赖，AimRT 支持 gcc 11.4 及以上的版本。

应该是灵犀X1的开源工程的问题，我去那边反馈吧。