AimRT/AimRT #77: log_util中log函数打印等级问题，容易出现段错误

flyonaie, 2024-11-03:
**Version of AimRT**
eg. v0.8.1

**Environments**
versions of gcc, cmake, python etc.

**Describe the bug**
A clear and concise description of what the bug is.
log_util中log函数打印等级问题，容易出现段错误
static constexpr uint32_t kLvlNameArraySize =
        sizeof(kLvlNameArray) / sizeof(kLvlNameArray[0]);
    if (lvl >= kLvlNameArraySize) lvl = kLvlNameArraySize;
应该改为
static constexpr uint32_t kLvlNameArraySize =
        sizeof(kLvlNameArray) / sizeof(kLvlNameArray[0]);
    if (lvl >= kLvlNameArraySize) lvl = kLvlNameArraySize - 1;

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Actual behavior**
A clear and concise description of what you have seen.

**Additional context**
Add any other context about the problem here.

ChuheZhang, 2024-11-03:
It appears there may be an error in the boundary check for the `lvl` variable in the logging function. Currently, the code sets `lvl` to `kLvlNameArraySize` if it exceeds the valid range:

```cpp
if (lvl >= kLvlNameArraySize) lvl = kLvlNameArraySize;
```

This could cause an out-of-bounds access since `kLvlNameArray[lvl]` would then be invalid, potentially leading to a segmentation fault. It seems that the intended behavior was likely to default `lvl` to `kLogLevelError` (or `kLogLevelFatal`) if it exceeds the defined range, rather than setting it to `kLvlNameArraySize`. This way, any unexpected log level would default to a high-priority level, like `Error` or `Fatal`, without risking out-of-bounds access.

Here’s the corrected line for improved robustness:

```cpp
if (lvl >= kLvlNameArraySize) lvl = kLogLevelError; // or kLogLevelFatal
```

Although in the current file `lvl` doesn’t actually exceed the `0-5` range, this adjustment would provide additional protection if `lvl` were to come from external sources in the future, such as configuration files or user input, preventing potential out-of-bounds issues.

ChuheZhang, 2024-11-03:
I agree with you.It appears there may be an error in the boundary check for the `lvl` variable in the logging function. Currently, the code sets `lvl` to `kLvlNameArraySize` if it exceeds the valid range:

```cpp
if (lvl >= kLvlNameArraySize) lvl = kLvlNameArraySize;
```

This could cause an out-of-bounds access since `kLvlNameArray[lvl]` would then be invalid, potentially leading to a segmentation fault. It seems that the intended behavior was likely to default `lvl` to `kLogLevelError` (or `kLogLevelFatal`) if it exceeds the defined range, rather than setting it to `kLvlNameArraySize`. This way, any unexpected log level would default to a high-priority level, like `Error` or `Fatal`, without risking out-of-bounds access.

Here’s the corrected line for improved robustness:

```cpp
if (lvl >= kLvlNameArraySize) lvl = kLogLevelError; // or kLogLevelFatal
```

Although in the current file `lvl` doesn’t actually exceed the `0-5` range, this adjustment would provide additional protection if `lvl` were to come from external sources in the future, such as configuration files or user input, preventing potential out-of-bounds issues.

PR #82 (2024-11-05): fix:  repair kLvlNameArray oversize
Close #77