AimRT/AimRT #122: 建议支持并简化本地化安装

vistar-terry, 2024-12-08:
目前使用 FetchContent 管理依赖，对于个人开发还好，但对于公司团队来说，大多是局域网本地化开发环境，而且是多人协同开发，每次新的编译都连接外网下载依赖源码不现实。
当然公司可以在局域网内部署依赖下载服务器，但这样复杂化了用户使用，不利于AimRT的推广。
建议执行一次安装AimRT，同时也把对应版本的三方依赖安装，或其他简化的支持本地化安装的方法。
祝AimRT快速发展成熟，被广泛应用，成为中国版ROS。

zhangyi1357, 2024-12-09:
非常感谢您的建议！

当前已经支持[源码编译安装](https://docs.aimrt.org/tutorials/quick_start/installation_cpp.html#find-package)，可能现在稍微麻烦的点就是需要自己 include 一些相关的依赖项

```
list(APPEND CMAKE_MODULE_PATH /path/to/aimrt/install/cmake)
include(GetFmt)
include(GetLibUnifex)
include(GetProtobuf)
include(GetYamlCpp)
include(GetJsonCpp)
include(GetTBB)
include(GetAsio)
```

不过我们在相应的 GetXXX.Fmt 中都设置了 XXX_DOWNLOAD_URL 和 XXX_LOCAL_SOURCE 变量，可以用来指定下载路径或者本地目录，通过相应的设置即可在内网进行开发。

flyonaie, 2025-03-02:
他的需求应该是
1、第一次安装下载源码后编译成sdk
2、第二次以及后面编译直接依赖sdk吧，这样就很方便

zhangyi1357, 2025-03-03:
> 他的需求应该是 1、第一次安装下载源码后编译成sdk 2、第二次以及后面编译直接依赖sdk吧，这样就很方便

你描述的这个需求我理解其实就是源码编译安装，现在已经支持了，只是 AimRT 还依赖一些其他库，这些库的内容的安装并没有被包含到 AimRT 的安装流程中，需要自行导入或者安装。