AimRT/AimRT #125: window10 环境 编译0.9.0  AimRT\src\examples\cpp\pb_chn\module\normal_subscriber_module 模块存在tuple越界问题

knight-cosmoon, 2024-12-12:
**Version of AimRT**
v0.9.0

**Environments**

 --Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.19045.
 --python.exe (found version "3.11.8")
 --cmake-3.31

**Describe the bug**

Visual Studio\set_up\IDE\VC\Tools\MSVC\14.35.32215\include\utility(538,19): error C2338: static_asser
t failed: 'tuple index out of bounds' [D:\AimRT\AimRT\build\src\examples\cpp\pb_chn\module\normal_subscriber_module\aim
rt_examples_cpp_pb_chn_normal_subscriber_module.vcxproj]

zhangyi1357, 2024-12-13:
无法复现该问题，复现环境：

- cmake version 3.29.5-msvc4
- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.19045.
- Python 3.11.9

请提供更多详细复现步骤。

knight-cosmoon, 2024-12-16:
--cmake version 3.31
--Python 3.11.8
禁用examples可以正常编译
编译环境: 国内 有vpn可以连外网
编译步骤如下:
1、pip install pyinstaller jinja2 pyyaml --upgrade
2、pip install build setuptools wheel --upgrade
3、修改build.bat 文件(打开python sdk编译) 详见txt:
[build.txt](https://github.com/user-attachments/files/18143363/build.txt)
4、cmd 打开命令台 进入aimrt工程目录: build.bat 执行编译

zhangyi1357, 2024-12-17:
按照给出步骤操作后仍然无法复现问题，无论是开启还是禁用 examples。

方便的话可以给出完整的编译日志进行进一步分析。

wtudio, 2024-12-17:
请升级编译器版本至 MSVC-1940或以上，然后重试

knight-cosmoon, 2024-12-17:
-MSVC版本更新后到1940以上后可以正常编译。
--Visual Studio 2022 的MSVC 工具集在2024年5月更新到了19.40版本， 早期版本低于1940 (建议官网文档：https://docs.aimrt.org/tutorials/quick_start/build_from_source_windows.html  MSVC编译套件位置描述稍微修改下) 

-0.9.1 有一个案例名称过长，需要修改注册表
--F:\My_Application\Visual Studio\set_up\IDE\MSBuild\Microsoft\VC\v170\Microsoft.CppBuild.targets(382,5): error MSB3491: 未能向文件“aimrt_examples_cpp_helloworld_helloworld_app_registration_mode.dir\Release\aimrt_ex.78780
62A.tlog\aimrt_examples_cpp_helloworld_helloworld_app_registration_mode.lastbuildstate”写入行。路径: aimrt_examples_cpp_helloworld_helloworld_app_registration_mode.dir\Release\aimrt_ex.7878062A.tlog\aimrt_examples_cpp_he
lloworld_helloworld_app_registration_mode.lastbuildstate 超过 OS 最大路径限制。完全限定的文件名必须少于 260 个字符。  [D:\AimRT\AimRT\build\src\examples\cpp\helloworld\app\helloworld_app_registration_mode\aimrt_examples_cpp_helloworld_hell
oworld_app_registration_mode.vcxproj]

closed 2024-12-17 by knight-cosmoon

zhangyi1357, 2024-12-20:
> -MSVC版本更新后到1940以上后可以正常编译。 --Visual Studio 2022 的MSVC 工具集在2024年5月更新到了19.40版本， 早期版本低于1940 (建议官网文档：https://docs.aimrt.org/tutorials/quick_start/build_from_source_windows.html MSVC编译套件位置描述稍微修改下)
>
> -0.9.1 [...]

感谢您的反馈，v0.9.2 bugfix 版本将会更新 msvc 套件具体描述。