AimRT/AimRT #134: win10 python3.11.8 protoc 生成服务定义部分的 Python 桩代码报错

knight-cosmoon, 2024-12-20:
**Version of AimRT**
eg. v0.9.1

**Environments**
win10
python3.11.8
libprotoc 29.1

**Describe the bug**
F:\Project\WorkSpacePython\aimrt_study\.venv\Lib\site-packages\aimrt_py>protoc --aimrt_rpc_out=. --plugin=protoc-gen-aimrt_rpc=. protoc_plugin_py_gen_aimrt_py_rpc.py rpc.proto
protoc_plugin_py_gen_aimrt_py_rpc.py:1:1: Expected top-level statement (e.g. "message").
protoc_plugin_py_gen_aimrt_py_rpc.py:19:136: Multiline strings are not allowed. Did you miss a "?.
protoc_plugin_py_gen_aimrt_py_rpc.py:185:4: Multiline strings are not allowed. Did you miss a "?.

zhangyi1357, 2024-12-20:
看起来你把插件当做了 proto 文件进行生成，考虑使用下面的命令

```
protoc --plugin=protoc-gen-aimrt_rpc=./protoc_plugin_py_gen_aimrt_py_rpc.py \
       --aimrt_rpc_out=. \
       rpc.proto
```

knight-cosmoon, 2024-12-20:
F:\Project\WorkSpacePython\aimrt_study\.venv\Lib\site-packages\aimrt_py>protoc --plugin=protoc-gen-aimrt_rpc=./protoc_plugin_py_gen_aimrt_py_rpc.py --aimrt_rpc_out=. rpc.proto
--aimrt_rpc_out: protoc-gen-aimrt_rpc: %1 is not a valid Win32 application.

F:\Project\WorkSpacePython\aimrt_study\.venv\Lib\site-packages\aimrt_py>protoc --plugin=protoc-gen-aimrt_rpc=. protoc_plugin_py_gen_aimrt_py_rpc.py --aimrt_rpc_out=. rpc.proto
protoc_plugin_py_gen_aimrt_py_rpc.py:1:1: Expected top-level statement (e.g. "message").
protoc_plugin_py_gen_aimrt_py_rpc.py:19:136: Multiline strings are not allowed. Did you miss a "?.
protoc_plugin_py_gen_aimrt_py_rpc.py:185:4: Multiline strings are not allowed. Did you miss a "?.

还是不行

zhangyi1357, 2024-12-20:
经过一些调查，这是因为 windows 无法将 python 文件作为可执行未见来执行，所以产生了这个报错。

> F:\Project\WorkSpacePython\aimrt_study.venv\Lib\site-packages\aimrt_py>protoc --plugin=protoc-gen-aimrt_rpc=./protoc_plugin_py_gen_aimrt_py_rpc.py --aimrt_rpc_out=. rpc.proto
> --aimrt_rpc_out: protoc-gen-aimrt_rpc: %1 is not a valid Win32 application.

不过安装 aimrt_py 后，aimrt_py 会为你生成一个等效于 `python.exe protoc_plugin_py_gen_aimrt_py_rpc.py` 的可执行文件名为 `protoc_plugin_py_gen_aimrt_py_rpc.exe`，位于你的 python 安装根目录的 Scripts 路径。

所以使用这个 exe 的完整命令应该为

```
protoc --plugin=protoc-gen-aimrt_rpc=/path/to/your/protoc_plugin_py_gen_aimrt_py_rpc.exe --aimrt_rpc_out=. rpc.proto
```

knight-cosmoon, 2024-12-20:
已解决，感谢

closed 2024-12-20 by knight-cosmoon