AimRT/AimRT #139: 建议提供一个 rpc模式下 单个client对应多个server 的交互案例

knight-cosmoon, 2024-12-27:
实际使用过程中有rpc模式单个client对应多个server的需求，examples中未看见对应案例，希望可以提供一个

zhangyi1357, 2024-12-27:
RPC 的概念就是一个单独的远程过程调用，是一对一的，框架侧暂时不支持直接调用一次请求调用多个 service 的功能，试想支持单个请求调用多个 service，其中部分成功部分失败的情况处理起来将会很复杂，与具体业务逻辑高度耦合。

不过如果真的存在这个场景，例如想要实现一个遥控指令同时操作多台机器人，每台机器人上的软件完全相同，service 的名称等均相同，可以考虑使用 http 后端，并在 context 中设置 AIMRT_RPC_CONTEXT_KEY_TO_ADDR 参数，然后利用并行手段达到同时调用的效果，具体可以参考 [RPC 配置文档](https://docs.aimrt.org/tutorials/cfg/rpc.html)：

> 当 Client 端接口层发起一个 RPC 请求后，AimRT 框架会根据以下规则，在多个 RPC 后端中选择一个进行实际的处理：
>
> 1. AimRT 框架会先根据clients_options配置确定某个 RPC 方法能被处理的RPC后端列表。
> 
> 2. AimRT 框架会先解析传入的 Contxet 里 Meta 参数中的AIMRT_RPC_CONTEXT_KEY_TO_ADDR项，如果其中手动配置了形如xxx://yyy,zzz这样的URL，则会解析出xxx字符串，并寻找同名的 RPC 后端进行处理。
> 
> 3. 如果没有配置 Contxet 参数，则根据该 RPC 方法能被处理的 RPC 后端列表顺序，依次尝试进行处理，直到遇到第一个真正进行处理的后端。

不过以上方法仍然无法避免部分成功部分失败等特殊情况的干扰，这是这种场景下业务逻辑自身的复杂性带来的。

knight-cosmoon, 2024-12-29:
不是一次请求调用多个service的功能哦，我这边是不太明白关于
  rpc:
    backends:
      - type: grpc
        options:
          clients_options:
            - func_name: "(.*)"
              server_url: http://127.0.0.1:50050

clients_options: 中短语 功能名与服务器的rul 对应关系的配置，主要疑问是：
-func_name: 配置的正则 是指命中proto中定义的服务函数名 就请求相对应的服务器对的吧？

zhangyi1357, 2024-12-30:
> clients_options下的-func_name: 字段是指的proto的service中配置的远程服务函数名称对吧，而不是service名称。

是的，更准确的名称可以通过服务端启动日志的初始化报告部分进行查询，如下：

```text
[11]. Rpc Server Info
+-------------------------------------------------------+-------------------------+----------+---------+
| func                                                  | module                  | backends | filters |
+-------------------------------------------------------+-------------------------+----------+---------+
| pb:/aimrt.protocols.example.ExampleService/GetBarData | NormalRpcCoServerModule | grpc     |         |
| pb:/aimrt.protocols.example.ExampleService/GetFooData | NormalRpcCoServerModule | grpc     |         |
+-------------------------------------------------------+-------------------------+----------+---------+
```

closed 2025-01-06 by AimRT