AimRT/AimRT #143: [Fea] cmake文件配置过度，模块之间编译过耦合。建议重新设计cmake方案

williamlzw, 2025-01-03:
cmake文件命名空间复杂。我的需求是只用到了自定义一个ros2 msg，并订阅。暂时用不到rpc，proto。
但是目前cmake整个架构得全部编译。能否解耦编译呢？比如先编译底层核心库(日志，线程协程)，其次选择通讯协议编译通讯库(ros2,rpc,proto,mqtt)，然后选择调用形式编译主程序(发布者，订阅者等)。额外功能单独编译如python包，监控，例子等。其中底层库Build目录固定，通讯库build目录可以固定，主程序目录用户指定。额外功能目录用户指定。
总体来说是支持需求最小化的cmake编译方式。建议重新设计cmake方案。

zhangyi1357, 2025-01-05:
感谢您的反馈！

AimRT 插件化的设计本身支持分离编译，所有插件以及 Python 包、ROS2 支持等均有相应编译选项可以关闭且在 cmake 中默认关闭，执行 build.sh 则会默认开启这些选项，可以修改 build.sh 来关闭这些选项。

至于 rpc、channel等功能接口层的分离编译在 AimRT 当前的代码规模下（从 0 构建在 intel i7@13700k 上开启 20 核编译仅需要数分钟时间）并没有太大的必要，且一般用户都会同时用到这些功能。另外更多的编译选项只会造成更多的困扰，正如这里提供了相应的插件以及 Python 等内容的分离编译选项但是作为用户的您仍不知情一样。

具体 CMake 相关的问题可以参考 [AimRT 中的 CMake 文档](https://docs.aimrt.org/tutorials/concepts/cmake.html)。

williamlzw, 2025-01-05:
新建代码工程，如何做到cmake最少配置就可以自定义一个ros2 msg，并订阅话题。代码工程跟Aimrt不是同一个目录下。

zhangyi1357, 2025-01-05:
一方面可以参考 AimRT 提供的 CLI 工具为项目生成脚手架代码：
https://docs.aimrt.org/tutorials/cli_tool/gen_prj.html

另一方面可以参考已有的基于 AimRT 的项目，使用类似的代码组织形式：
https://github.com/AgibotTech/agibot_x1_infer

BTW，AiMRT 在 Windows 下对 ROS2 的支持未经测试，如果要使用 ROS2 msg 建议基于 Linux 平台进行。

zhangyi1357, 2025-01-05:
AimRT从入门到精通（七） ：AimRT中的Channel-自定义消息类型及后端 - 蒋程扬的文章 - 知乎
https://zhuanlan.zhihu.com/p/2558246320

以上教程也是一个不错的参考信息源。

williamlzw, 2025-01-05:
windows版本能否提供visual studio 2022 sln解决方案形式的aimrt工程，可以开个Windows分支。直接打开sln编译，不需要cmake编译。第三方依赖建议直接提供预编译lib或dll文件，代码多的不要用源码形式编译第三方（尤其是boost），代码少的如yaml-cpp可以集成到sln里面。

zhangyi1357, 2025-01-05:
> windows版本能否提供visual studio 2022 sln解决方案形式的aimrt工程，可以开个Windows分支或者AimRT-WINDOWS仓库。直接打开sln编译，不需要cmake编译。第三方依赖建议直接提供预编译lib或dll文件，代码多的不要用源码形式编译第三方（尤其是boost），代码少的如yaml-cpp可以集成到sln里面。

感谢您的建议！

目前的跨平台支持依赖于 CMake 和 C++ 自身的跨平台支持，单独维护一套 VS 工程对于目前的开发资源来说较为吃紧。同样地，因为资源问题以及使用人数，Windows 平台暂时不是主要支持的目标平台，Windows 下开发可能需要开发者自身较为熟悉 CMake 以及 Windows 开发流程，如果您有兴趣可以考虑新建一个 AimRT-Windows 仓库，如果仓库基本可用可以列入开发文档中。

closed 2025-01-06 by AimRT