AimRT/AimRT #144: windows编译了normal_subscriber_app.exe和normal_publisher_app.exe，同时运行为何不打印消息

williamlzw, 2025-01-05:
visual studio2022编译了aimrt基于runtime库，包括normal_subscriber_app.exe和normal_publisher_app.exe。
[image]

这两个不是配套的订阅者和发布者吗？同时运行exe控制台为何不打印收到的信息？
[image]
[image]

williamlzw, 2025-01-05:
传入yaml配置文件报错重复模块
[image]
[image]

zhangyi1357, 2025-01-05:
不清楚你的态度是认为这是一个 bug 还是单纯求助，看起来有两个问题，不知道哪个更重要，同时还带上了 enhancement 标签，让人有些迷惑。

但是无论如何，请遵循 bug report 模板，提供详细的复现步骤，很乐意为你解决问题，前提是一个清晰且明确的问题而不是几张截图。

williamlzw, 2025-01-05:
按照勾选第一张图配置，我用cmake 3.31 GUI可视化生成visual studio2022的sln解决方案。进入sln工程编译all_build 64位。在Debug生成文件夹有normal_subscriber_app.exe和normal_publisher_app.exe。控制台进入Debug目录，./normal_publisher_app.exe  test_cfg.yaml运行报错重复模块，yaml文件内容如最后一张图所示。
Debug目录文件如图
[image]

williamlzw, 2025-01-05:
我将几个主要文件拷出来，运行单个exe还是报错。如图运行normal_publisher_app.exe，带参运行examples_cpp_pb_chn_cfg.yaml。
[image]

[image]

zhangyi1357, 2025-01-05:
将配置中这一段删除：

[image]

publisher_app 采用 app 模式在 main 函数中为 core 注册了一个 NormalPublisherModule，在配置文件中配置以上内容会另外导入一次单独编写的 NormalPublisherModule，从而引发上述报错。

关于 channel 的样例如何运行可以参考 https://github.com/AimRT/AimRT/blob/main/src/examples/cpp/pb_chn/README.md，启动脚本是为 linux 准备的，但是配置和原理都是通用的。

另外，如果不理解 Pkg 模式和 App 模式，可以参考 https://docs.aimrt.org/tutorials/concepts/concepts.html#id2

williamlzw, 2025-01-05:
按照上图我删除了pkgs节点，运行发布者可以打印发布的消息。但是运行normal_subscriber_app.exe能启动，但是不打印消息。
[image]

zhangyi1357, 2025-01-05:
看一下上面我提供的 examples 的链接，这两个不是配套的，每一个都是一个单独的例子，因为这里使用的最基础的 local 后端只能用于同进程通信。

如果要启动两个单独的程序用于一收一发，需要结合通信插件，最基础的可以了解网络插件提供的 http 后端，示例说明：https://github.com/AimRT/AimRT/blob/main/src/examples/plugins/net_plugin/README.md#http-protobuf-channel

williamlzw, 2025-01-05:
[image]

这个ros2_chn也只能同进程用ros2话题方式进行通讯？有没有与原生ros2写的进程订阅话题的例子？

zhangyi1357, 2025-01-05:
> ![image](https://private-user-images.githubusercontent.com/11348670/400212116-115f93b2-be7c-4880-97e7-e30a56a89b0d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

是不是只能同进程通信这一点只和使用什么后端有关，local 后端只提供同进程通信能力。
使用 ROS2 插件即可实现远程通信并且对接原生 ROS2 节点的能力，示例可以参考：
https://github.com/AimRT/AimRT/tree/main/src/examples/plugins/ros2_plugin#ros2-channel

其实这些信息只需要翻看一下官方文档即可找到，建议仔细浏览一下官方文档：
https://docs.aimrt.org/tutorials/plugins/ros2_plugin.html

closed 2025-01-05 by williamlzw