AimRT/AimRT #146: aimrt中的grpc服务函数，如何在框架外部通过grpc进行远程调用？

knight-cosmoon, 2025-01-06:
想与第三方框架结合，但是用标准grpc插件生成的包无法正常调用框架内部的函数，会报如下错误：
@aimrt::plugins::grpc_plugin::server::Connection::Start::<lambda_1>::operator ()]Http svr session get bad request, remote addr 127.0.0.1, url not registered: /service.arm.ArmService/get_actual_joint
请问有什么方法可以不进行注册验证吗？

zhangyi1357, 2025-01-07:
这个 register 说的不是 网页/app 概念的登录验证，而是类似于访问网址 ${host}:${port}/service.arm.ArmService/get_actual_joint，然后提示网址未注册。

这个问题应该是在 https://github.com/AimRT/AimRT/pull/81 中引入的与原生 grpc 通信的一个 bug，其中将注册的网址从 ${host}:${port}/service.arm.ArmService/get_actual_joint 修改成了 ${host}:${port}/rpc/service.arm.ArmService/get_actual_joint。

临时解决方案：
修改客户端请求的网址，将其加上 /rpc 前缀，例如针对 python 可以修改生成的 xxx_pb2_grpc.py 文件，在如下所示位置添加 /rpc 前缀
[image]
 
后续我会提交一个 PR 修复这个路径问题，只需要更新框架即可正常与原生 grpc 通信。

zhangyi1357, 2025-01-07:
另外目前在 examples 中提供了与原生python grpc 进行通信的示例，可以参考：
https://github.com/AimRT/AimRT/tree/main/src/examples/plugins/grpc_plugin

PR #149 (2025-01-07): fix: correct service address handling in grpc plugin
- Fixed an issue in the grpc plugin where the service address was incorrectly set during communication with native grpc. This change ensures that the service address is now correctly formatted, improving the reliability of grpc interactions.
- Updated the release notes for version 0.10.0 to reflect this fix.

Close #146 .
closed 2025-01-07 by AimRT