AimRT/AimRT #157: [Fea] src目录下面examples目录提到跟src同级。build目录文件位置改进

williamlzw, 2025-01-09:
目前src/examples目录是例子，跟框架src同级引起混乱。build目录生成也是库文件和
期望:
1、examples目录跟src目录同级
2、生成的examples exe文件和依赖单独放在build目录下的examples目录下(build/examples)
3、同理，install目录也区分库so文件和examples例子。
4、build目录.a文件建议用文件夹单独存放。

zhangyi1357, 2025-01-10:
感谢您的反馈！

问题 1 和问题 2 的目录结构调整属于个人偏好，目前的结构已经很清晰且运行良好。

问题 3 不存在，install 目录不会包含 examples。

问题 4 在 install 中 .a 已经是单独存放于 lib 中，build 目录下无整理必要，本身程序运行也不使用这些东西，安装只需要移动 install 目录即可。

williamlzw, 2025-01-10:
examples目录写成脚手架生成的代码形式，演示多个module构成单个脚手架项目，单个pkg构成脚手架项目，引入外部cpp文件脚手架等等

williamlzw, 2025-01-10:
脚手架GetAimrt.cmake有点复杂了。演示如何直接引用源码build/install目录下的bin so protocols目录文件。 比如我的做法是先在src下面protocols增加需要用到的协议文件和aimrt整体编译。做项目时脚手架希望直接引用build/install的文件，不需要脚手架项目里额外protocols文件夹重新编译协议文件。

zhangyi1357, 2025-01-10:
> examples目录写成脚手架生成的代码形式，演示多个module构成单个脚手架项目，单个pkg构成脚手架项目，引入外部cpp文件脚手架等等

如何源码引用 AimRT 和 AimRT 本身的 examples 是两码事，需要自己新建项目的情况下理应自行处理自己项目的目录结构，与 AimRT 自身的 examples 如何组织没什么关系，将 AiMRT 作为一个依赖性自行构建即可。如果不熟悉 CMake 构建的话可以参考 [X1 项目](https://github.com/AgibotTech/agibot_x1_infer)的组织形式。

> 脚手架GetAimrt.cmake有点复杂了。演示如何直接引用源码build/install目录下的bin so protocols目录文件。 比如我的做法是先在src下面protocols增加需要用到的协议文件和aimrt整体编译。做项目时脚手架希望直接引用build/install的文件，不需要脚手架项目里额外protocols文件夹重新编译协议文件。

自己项目的自定义 protocol 不应放在 AimRT 主仓的 protocol 目录中，自己项目的内容应该自行管理，上面 X1 项目中也有自定义协议，可以自己参考编写相应 CMake 文件进行构建。

williamlzw, 2025-01-11:
现在代码架构编译ros2消息类型，命名空间跟目录结构强相关，由CMakeLists.txt定义。如果每个项目protocols文件夹不同层级结构，导致编译出来的rosidl文件不好统一命名空间。所以才建议protocols放到跟框架一起编译。

williamlzw, 2025-01-11:
examples生成到Build目录文件过多，混乱。建议examples单独生成到一个目录。

zhangyi1357, 2025-01-11:
> 现在代码架构编译ros2消息类型，命名空间跟目录结构强相关，由CMakeLists.txt定义。如果每个项目protocols文件夹不同层级结构，导致编译出来的rosidl文件不好统一命名空间。所以才建议protocols放到跟框架一起编译。

自己项目的 protocol 跟放框架里不是合理的架构，不会为此提供任何支持。每个项目都应自行处理自己的 protocol 相关的内容，将不再对此进行讨论和支持。

closed 2025-01-13 by AimRT