AimRT/AimRT #174: 用标准grpcio.aio库 实现的客户端 无法实现对aimrt grpc服务器接口的异步请求

knight-cosmoon, 2025-01-17:
已知
1、用aimrt内置的客户端可以异步请求
2、客户端启用两个asyncio事件循环可以异步请求

问题:
在同一个asyncio事件循环中 当有一个stub与aimrt客户端进行通讯并等待响应时，客户端无法执行其他aimrt接口调用

zhangyi1357, 2025-01-17:
可以麻烦提供一个最小示例吗？

knight-cosmoon, 2025-01-18:
[aimrt_grpc_test.zip](https://github.com/user-attachments/files/18463526/aimrt_grpc_test.zip)

[ReadMe.txt](https://github.com/user-attachments/files/18463529/ReadMe.txt)

zhangyi1357, 2025-01-21:
感谢您的反馈！

首先目前 aimrt_py 的 rpc 并不支持异步的接口，所以无法注册 async 函数，只能注册普通函数。

但是这个问题是客户端（grpc.aio）发起的异步请求，理论上服务端有多个线程（在 grpc_plugin 配置项的 thread_num 中配置），即使在 service 中调用了 `time.sleep` 也应该正确调度到多个线程中并发执行，所以这是一个 bug。

问题调查过程：

- 通过在多个进程同时请求这两个 service 函数，现象是可以并行运行两个 service 函数，这说明问题出在同一条 http2 连接上的出现多个请求时

- 经过 wiresharek 抓包调查，有问题的代码指向了

https://github.com/AimRT/AimRT/blob/63d37cf97f28f38de0c758f850c5cee31e73e9e4/src/plugins/grpc_plugin/server/connection.h#L86-L108

- 以上代码中一次接收到多个请求时，在同一个 `socket_strand_` 中串行执行 service 函数，我已提交了一个 PR #176  修复此问题，您可以等待 PR 合并后更新代码

knight-cosmoon, 2025-01-21:
收到，感谢您的回复，期待问题早日修复。

PR #176 (2025-01-22): refactor: enhance request handling and response submission in gRPC connection  
- Improved the handling of incoming requests by checking the size of the request list and sending appropriate responses when no requests are present.
- Introduced a new `SubmitAndSend` method to streamline the submission of responses to the nghttp2 session, enhancing code clarity and maintainability.
- Updated error handling to log warnings for unregistered URLs and ensure proper response generation for bad requests.
- Refactored the request processing loop to utilize `asio::co_spawn` for concurrent handling of req [... cut]
closed 2025-01-22 by AimRT