AimRT/AimRT #182: 动态库加载问题： 模块是通过动态库加载的（libpb_rpc_pkg.so） 在动态库中使用 std::cout 可能会遇到符号解析或初始化问题

flyonaie, 2025-02-05:
**Version of AimRT**
应该都存在

**Environments**
- gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.2)  因为我将其改为了C++17编译来适应自己的项目
- cmake
- 动态库环境

**Describe the bug**
在动态库中使用 std::cout 导致段错误。具体表现为在动态库（libpb_rpc_pkg.so）中使用 std::cout 时出现 SIGSEGV 段错误。具体内容见最后的附录；

**To Reproduce**
1. 在动态库模块中使用 std::cout 进行打印
2. 运行程序后出现段错误
3. 错误堆栈显示在 std::ostream::sentry::sentry(std::ostream&) 处崩溃

**Expected behavior**
动态库中的 std::cout 语句能够正常执行，不会导致程序崩溃。

**Actual behavior**
程序在执行到动态库中的 std::cout 语句时发生段错误，堆栈信息显示在 libstdc++.so.6 中的 std::ostream::sentry::sentry 处崩溃。

**Additional context**
解决方案：

1. 修改动态库加载方式（推荐）：
   - 将 `dlopen` 的参数从 `RTLD_NOW | RTLD_LOCAL | RTLD_DEEPBIND` 修改为：
     - 方案1：`RTLD_LAZY | RTLD_GLOBAL`
     - 方案2：`RTLD_NOW | RTLD_LOCAL`

2. 参数说明：
   - RTLD_NOW vs RTLD_LAZY:
     - RTLD_NOW: 立即解析所有符号
     - RTLD_LAZY: 延迟解析符号（加载更快，但可能在运行时出现未定义符号错误）
   
   - RTLD_LOCAL vs RTLD_GLOBAL:
     - RTLD_LOCAL: 符号仅本库可见
     - RTLD_GLOBAL: 符号全局可见（可解决 std::cout 等全局符号共享问题）

   - RTLD_DEEPBIND:
     - 优先使用本库符号
     - 可能导致 std::cout 实例不一致问题

其他尝试过但无效的方案：
1. 强制动态链接 C++ 标准库：
```cmake
target_link_libraries(libpb_rpc_pkg PUBLIC -static-libgcc -static-libstdc++)
```

2. 设置特殊编译选项：
```cmake
target_compile_options(${CUR_TARGET_NAME} PRIVATE
    -fPIC
    -rdynamic
    -fvisibility=hidden
)

set_target_properties(${CUR_TARGET_NAME} PROPERTIES
    LINK_FLAGS "-Wl,--no-as-needed"
)

---------------------------------------------------------------------------------------------------
附录：
### 后台打印如下：
```
/channel_manager.cc:270:0 @Start]Channel manager start completed.
111111111111ModuleManager::Start = 2
222222222222ModuleManager::module_name = NormalRpcSyncClientModule
333333333333ModuleManager::module_ptr-impl = 0x5555555b5b60
333333333333ModuleManager::module_ptr-start = 0x7ffff49534ce

Thread 1 "aimrt_main" received signal SIGSEGV, Segmentation fault.
0x00007ffff767adba in std::ostream::sentry::sentry(std::ostream&) () from /lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) bt full
#0  0x00007ffff767adba in std::ostream::sentry::sentry(std::ostream&) () from /lib/x86_64-linux-gnu/libstdc++.so.6
No symbol table info available.
#1  0x00007ffff767b8dc in std::basic_ostream<char, std::char_traits<char> >& std::__ostream_insert<char, std::char_traits<char> >(std::basic_ostream<char, std::char_traits<char> >&, char const*, long) ()
   from /lib/x86_64-linux-gnu/libstdc++.so.6
No symbol table info available.
#2  0x00007ffff767bddb in std::basic_ostream<char, std::char_traits<char> >& std::operator<< <std::char_traits<char> >(std::basic_ostream<char, std::char_traits<char> >&, char const*) () from /lib/x86_64-linux-gnu/libstdc++.so.6
No symbol table info available.
#3  0x00007ffff4955d4c in aimrt::examples::cpp::pb_rpc::normal_rpc_sync_client_module::NormalRpcSyncClientModule::Start
    (this=0x5555555b5b60)
    at /usr/local/workspace/github/aimrt_all/OmniBot/omnirt/src/examples/cpp/pb_rpc/module/normal_rpc_sync_client_module/normal_rpc_sync_client_module.cc:45
        __FUNCTION__ = "Start"
#4  0x00007ffff49534a8 in aimrt::ModuleBase::GenBase(void*)::{lambda(void*)#3}::operator()(void*) const (
    __closure=0x0, impl=0x5555555b5b60)
    at /usr/local/workspace/github/aimrt_all/OmniBot/omnirt/src/interface/aimrt_pkg_c_interface/../aimrt_module_cpp_interface/module_base.h:88
No locals.
#5  0x00007ffff49534ef in aimrt::ModuleBase::GenBase(void*)::{lambda(void*)#3}::_FUN(void*) ()
    at /usr/local/workspace/github/aimrt_all/OmniBot/omnirt/src/interface/aimrt_pkg_c_interface/../aimrt_module_cpp_interface/module_base.h:92
No locals.
#6  0x00007ffff7e39afa in aimrt::runtime::core::module::ModuleManager::Start (this=0x7fffffffe128)
--Type <RET> for more, q to quit, c to continue without paging--RET
    at /usr/local/workspace/github/aimrt_all/OmniBot/omnirt/src/runtime/core/module/module_manager.cc:244
        module_wrapper_map_itr = {<std::__detail::_Node_iterator_base<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::unique_ptr<aimrt::runtime::core::module::ModuleManager::ModuleWrapper, std::default_delete<aimrt::runtime::core::module::ModuleManager::ModuleWrapper> > >, true>> = {
            _M_cur = 0x5555555bf910}, <No data fields>}
        module_ptr = 0x5555555b5b68
        start_ret = false
        module_name = "NormalRpcSyncClientModule"
        __for_range = std::vector of length 2, capacity 2 = {"NormalRpcSyncClientModule", "NormalRpcSyncServerModule"}
        __for_begin = "NormalRpcSyncClientModule"
        __for_end = <error: Cannot access memory at address 0x641>
        __FUNCTION__ = "Start"
#7  0x00007ffff7ccb410 in aimrt::runtime::core::AimRTCore::StartImpl (this=0x7fffffffd380)
    at /usr/local/workspace/github/aimrt_all/OmniBot/omnirt/src/runtime/core/aimrt_core.cc:187
        __FUNCTION__ = "StartImpl"
#8  0x00007ffff7ccb753 in aimrt::runtime::core::AimRTCore::Start (this=0x7fffffffd380)
    at /usr/local/workspace/github/aimrt_all/OmniBot/omnirt/src/runtime/core/aimrt_core.cc:255
        __FUNCTION__ = "Start"
#9  0x0000555555558cd1 in ?? ()
No symbol table info available.
#10 0x00007ffff733d083 in __libc_start_main (main=0x555555558a82, argc=2, argv=0x7fffffffe358, init=<optimized out>, 
    fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffe348) at ../csu/libc-start.c:308
        self = <optimized out>
        result = <optimized out>
--Type <RET> for more, q to quit, c to continue without paging--RET
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {93824992256976, -274748435217756211, 93824992249632, 
                140737488347984, 0, 0, 274748436104892365, 274732390703583181}, mask_was_saved = 0}}, priv = {pad = {
              0x0, 0x0, 0x2, 0x7fffffffe358}, data = {prev = 0x0, cleanup = 0x0, canceltype = 2}}}
        not_first_call = <optimized out>
#11 0x000055555555874e in ?? ()
No symbol table info available.
```
### 错误代码如下
···
bool NormalRpcSyncClientModule::Initialize(aimrt::CoreRef core) {
  core_ = core;
  AIMRT_INFO("NormalRpcSyncClientModule::Initialize starting...");
 
  try {
    // Read cfg
    std::string file_path = std::string(core_.GetConfigurator().GetConfigFilePath());
    if (!file_path.empty()) {
      YAML::Node cfg_node = YAML::LoadFile(file_path);
      rpc_frq_ = cfg_node["rpc_frq"].as<double>();
    }

    // Get executor handle
    executor_ = core_.GetExecutorManager().GetExecutor("work_thread_pool");
    AIMRT_CHECK_ERROR_THROW(executor_, "Get executor 'work_thread_pool' failed.");

    // Get rpc handle
    auto rpc_handle = core_.GetRpcHandle();
    AIMRT_CHECK_ERROR_THROW(rpc_handle, "Get rpc handle failed.");

    // Register rpc client
    bool ret = aimrt::protocols::example::RegisterExampleServiceClientFunc(rpc_handle);
    AIMRT_CHECK_ERROR_THROW(ret, "Register client failed.");

  } catch (const std::exception& e) {
    AIMRT_ERROR("Init failed, {}", e.what());
    return false;
  }

  // 错误使用引起的冲突
  std::cout << "Start...111111111111" << std::endl;
  AIMRT_INFO("NormalRpcSyncClientModule Init succeeded.");
  return true;
}
···

zhangyi1357, 2025-02-05:
在 v0.9.2 版本中测试无能成功复现此问题。麻烦问一下您使用的是什么版本的 AimRT？

这个问题是曾经遇到过的问题，在 #43 提交后，确实观察到了无法在动态库的 module 中使用 std::cout 的问题，但是在 #120 中该问题得到了修复。

flyonaie, 2025-02-05:
1、版本我忘记了，应该是0.9.1，因为我改成了自己的版本号了
2、我感觉这个应该不是必现的，我现在在源码里面改了比较多，在主线程中加了比较多std::cout，所以出现了

flyonaie, 2025-02-05:
您好，有没有aimrt内部架构的讨论群？

zhangyi1357, 2025-02-06:
> 您好，有没有aimrt内部架构的讨论群？

暂时没有，有关于 aimrt 架构相关的讨论可以直接新开一个 issue，活跃开发者看到都会回复的。

AimRT, 2025-02-06:
> 您好，有没有aimrt内部架构的讨论群？

您可以参考[contact](https://docs.aimrt.org/contact/index.html)文档，技术问题可以直接开issue，其他问题可以通过邮箱与我们联系。

zhangyi1357, 2025-02-06:
> 1、版本我忘记了，应该是0.9.1，因为我改成了自己的版本号了 2、我感觉这个应该不是必现的，我现在在源码里面改了比较多，在主线程中加了比较多std::cout，所以出现了

可以将你的主线程（或许应该称为 main 可执行文件的编译单元）的编译选项加上 -fPIC，这样也可以解决此问题。

你所建议的移除 RTLD_DEEPBIND 是导致这个问题的根源，正在评估考虑一个较优的方式修复此问题。

具体的原因可以参考：http://blog.gerryyang.com/gcc/clang/2021/02/13/elf-plt-hook.html#rtld_deepbind-dlopen

PR #187 (2025-02-07): fix: simplify dynamic library loading and improve CMake PIC configuration
- Remove `RTLD_DEEPBIND` flag when loading dynamic libraries to prevent potential singleton mode disruption
- Replace manual `-fPIC` compiler flags with `CMAKE_POSITION_INDEPENDENT_CODE` for more robust PIC handling across compilers

Close #182 .
closed 2025-02-07 by AimRT