AimRT/AimRT #196: AimRT 中的 “Node” 概念，在代码中哪里体现的，我得场景是支持单个进程，也可以支持多个进程部署

flyonaie, 2025-02-13:
AimRT 中的 “Node” 概念[](https://docs.aimrt.org/tutorials/concepts/concepts.html#aimrt-node)
Node代表一个可以部署启动的进程，在其中运行了一个 AimRT 框架的 Runtime 实例。Node是一个部署、运行层面的概念，一个Node中可能存在多个Module。Node在启动时可以通过配置文件来设置日志、插件、执行器等运行参数。
这个在代码中咋体现的，是如何通过配置文件实现的

zhangyi1357, 2025-02-13:
Node 不是代码层面的概念，你可以理解为每启动一个 AimRT runtime 进程就是一个 Node，其实就是操作系统中的一个进程。

一般来说一个 AimRT Runtime 进程就会对应一份配置文件。

flyonaie, 2025-02-14:
一套软件包含三个模块a,b,c,三个在一个进程内运行，然后也希望a,b在一个进程运行，c在另一个进程中运行，这个不希望改源码，可以按照这两种形式部署

zhangyi1357, 2025-02-14:
可以直接把三个 module 放到一个 pkg 里面，然后写 3 份配置，一份加载 abc，另外两份分别加载 ab 和 c。

可以参考以下写法

https://github.com/AimRT/AimRT/blob/f5d33ff98c90bb557b08c772c82972584ab340f2/src/examples/cpp/pb_rpc/install/linux/bin/cfg/examples_cpp_pb_rpc_benchmark_single_pkg_cfg.yaml#L38-L46

closed 2025-02-17 by zhangyi1357