AimRT/AimRT #223: aimrt_grpc服务端在消息较大时，不响应python grpcio库编写的客户端，但是可以响应aimrt_grpc编写的客户端

knight-cosmoon, 2025-03-13:
传输图片字节流时，640*480分辨率的图片可以正常传输到python grpcio库编写的客户端，1280*720分辨率的图片无法正常传输(wireshark 抓包显示服务器未响应数据)；
传输1280*720分辨率的图片时，python grpc客户端报错信息如下：

Traceback (most recent call last):
  File "D:\code_repository\python_project\aimrt_robot\src\app\fastapi\aimrt_grpc_package\real_service\device\head_camera_client.py", line 65, in <module>
    asyncio.run(main())
  File "D:\software\design\python\environment\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "D:\software\design\python\environment\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\software\design\python\environment\Python311\Lib\asyncio\base_events.py", line 654, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "D:\code_repository\python_project\aimrt_robot\src\app\fastapi\aimrt_grpc_package\real_service\device\head_camera_client.py", line 46, in main
    result = await asyncio.wait_for(obj.get_color_image(), timeout=30)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\software\design\python\environment\Python311\Lib\asyncio\tasks.py", line 489, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "D:\code_repository\python_project\aimrt_robot\src\app\fastapi\aimrt_grpc_package\real_service\device\head_camera_client.py", line 34, in get_color_image
    res = await self.__rpc(req_dict=req_dict, func=self.stub.get_color_image)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\code_repository\python_project\aimrt_robot\src\app\fastapi\aimrt_grpc_package\real_service\device\head_camera_client.py", line 30, in __rpc
    return await func(request)
           ^^^^^^^^^^^^^^^^^^^
  File "D:\code_repository\python_project\aimrt_robot\.venv\Lib\site-packages\grpc\aio\_call.py", line 327, in __await__
    raise _create_rpc_error(
grpc.aio._call.AioRpcError: <AioRpcError of RPC that terminated with:
	status = StatusCode.UNKNOWN
	details = "Stream removed"
	debug_error_string = "UNKNOWN:Error received from peer ipv4:127.0.0.1:50050 {created_time:"2025-03-13T07:06:21.9501325+00:00", grpc_status:2, grpc_message:"Stream removed"}"
>

zhangyi1357, 2025-03-18:
这个能方便提供一个可复现示例吗

zhangyi1357, 2025-03-18:
我这边复现现象反而相反，太大的包 aimrt_grpc 客户端会无法正确传输， grpcio 所写的客户端可以正确发送。

超过一定的大小会超过 grpcio 默认设置的大小报错如下：

```
GetFooData response: echo Hello Foo
GetBarData response: RPC failed. error code: StatusCode.RESOURCE_EXHAUSTED, error message: CLIENT: Received message larger than max (10000010 vs. 4194304)
```

knight-cosmoon, 2025-03-19:
[aimrt_grpc_test_image.zip](https://github.com/user-attachments/files/19335792/aimrt_grpc_test_image.zip)

[ReadMe.txt](https://github.com/user-attachments/files/19335796/ReadMe.txt)

zhangyi1357, 2025-03-19:
> [aimrt_grpc_test_image.zip](https://github.com/user-attachments/files/19335792/aimrt_grpc_test_image.zip)
> 
> [ReadMe.txt   自述文件.txt](https://github.com/user-attachments/files/19335796/ReadMe.txt)

已经在用你提供的示例验证了刚刚提交的 #233 可以修复此问题，如下图所示。

[image]

zhangyi1357, 2025-03-19:
另外已经测试过原生 grpcio 和 aimrt grpc 客户端通过本地回环向回显服务端发送 1 GB 的数据，修复后是可以正常接收和发送的。

PR #233 (2025-03-19): fix(grpc_plugin): send big data chunks until ending in the client side
Ref: https://github.com/nghttp2/nghttp2/issues/1046#issuecomment-341863998

Close #223 .
closed 2025-03-19 by wtudio