AimRT/AimRT #226: [Bug] grpc后端 周期性出现通讯延时

knight-cosmoon, 2025-03-14:
**Version of AimRT**
eg. v0.10.0-rc1

**Environments**
python 3.11.8;  windows 11

**Describe the bug**
aimrt grpc后端实现的 服务器与客户端通讯，当客户端连续请求服务端函数时，会在1min左右出现周期性通讯超时

knight-cosmoon, 2025-03-14:
[ReadMe.txt](https://github.com/user-attachments/files/19239635/ReadMe.txt)
[aimrt_grpc_test.zip](https://github.com/user-attachments/files/19239636/aimrt_grpc_test.zip)

最小测试案例如上：

zhangyi1357, 2025-03-14:
感谢您的反馈，等我调查一下给您一个答复。

另外问一下这个 issue 和 https://github.com/AimRT/AimRT/issues/223 是相关的吗？

knight-cosmoon, 2025-03-14:
不是同一个问题吧，之前那个问题是传输数据太大，标准的python grpc库实现的客户端接收不到服务器响应。但是用aimrt_grpc实现的客户端能够接收到较大数据, 测试的图片大小约1M，转bytes传的。

zhangyi1357, 2025-03-18:
最近有些忙，不过调查后这个问题较为简单，仅仅是忘记在收到请求时将连接上的数据标志（`tick_has_data_`）设为 `true`，从而导致服务端一直认为未收到数据断开与客户端的连接。

我已提交了一个 PR #230  修复此问题。

PR #230 (2025-03-18): fix(grpc_plugin): set tick_has_data_ to true when received request
Close #226 .
closed 2025-03-18 by wtudio