AimRT/AimRT #281: 录播插件支持ros2消息类型优化建议

luojinghua50, 2025-04-29:
在使用aimrt TAG v0.9.1 的录播插件，开发对机器人动作进行录制的需求，遇到一些问题。

需求背景：
在智元agibot-x1-infer 项目中进行二次开发，引入录播插件，yaml配置了
          type_support_pkgs:          
            - path: ./libexample_event_ts_pkg.so 
                topic_meta_list:
                  - topic_name: /joint_cmd  #关节指令topic
                    msg_type: my_ros2_proto::msg::JointCommand #ros2 消息类型
                    serialization_type: ros2 # optional 
启动项目报错，发现是libexample_event_ts_pkg.so  不支持消息类型my_ros2_proto::msg::JointCommand

定位到aimrt 的src/examples/plugins/record_playback_plugin/example_event_ts_pkg/type_support_pkg_main.cc 下只定义了
ExampleEventMsg 和 RosTestMsg 消息类型。

建议：是否可以支持ros2 自带的消息类型进来呢，而不需要我再这边去定义消息类型属性。

closed 2025-05-08 by wtudio

luojinghua50, 2025-05-09:
@yglsaltfish 你好，我看到你在src/protocols/ros2/ros2_msgs/type_support_pkg_main.cc 下新增了ros2支持的消息类型，对此我在X1 引用最新aimrt，并且调整了yaml配置如下：
      - name: record_playback_plugin
        path: ./libaimrt_record_playback_plugin.so
        options:
          timer_executor: storage_executor  # require time schedule!
          type_support_pkgs:       
            - path: ./libros2_msgs_ros2_ts.so 
          record_actions:
            - name: my_signal_record
              options:
                bag_path: ./record
                mode: signal # imd/signal
                max_preparation_duration_s: 10 # Effective only in signal mode
                executor: record_thread # require thread safe!
                storage_policy:
                  storage_format: mcap
                  max_bag_size_m: 2048
                  msg_write_interval: 1000        # message count period
                  msg_write_interval_time: 1000   # ms
                  journal_mode: WAL
                  synchronous_mode: full
                ext_data:
                  - key: platform
                    value: arm64
                  - key: os
                    value: ubuntu
                topic_meta_list:
                  - topic_name: /joint_states
                    msg_type: sensor_msgs::msg::JointState
                    serialization_type: ros2 # optional
其中 type_support_pkgs:       
           - path: ./libros2_msgs_ros2_ts.so 

                topic_meta_list:
                  - topic_name: /joint_states
                    msg_type: sensor_msgs::msg::JointState
                    serialization_type: ros2 # optional

启动运行X1项目时报错，找不到record_playback_plugin 支持该消息类型
[2025-05-09 14:38:28.375723][Error][421825][/home/qudian/robot/ljh/agibot-x1-infer/build/_deps/aimrt-src/src/plugins/record_playback_plugin/record_action.cc:151 @Initialize]Can not find type 'sensor_msgs::msg::JointState' in any type support pkg!
[2025-05-09 14:38:28.375751][Error][421825][/home/qudian/robot/ljh/agibot-x1-infer/build/_deps/aimrt-src/src/plugins/record_playback_plugin/record_playback_plugin.cc:249 @Initialize]Initialize failed, Can not find type 'sensor_msgs::msg::JointState' in any type support pkg!
[2025-05-09 14:38:28.375757][Error][421825][/home/qudian/robot/ljh/agibot-x1-infer/build/_deps/aimrt-src/src/runtime/core/plugin/plugin_manager.cc:118 @Initialize]Init plugin 'record_playback_plugin' failed.

请问这个是为什么？需要做什么调整呢，期待得到你的答复。

yglsaltfish, 2025-05-11:
可以参考https://github.com/AimRT/AimRT/blob/main/src/examples/plugins/record_playback_plugin/install/linux/bin/cfg/examples_plugins_record_playback_plugin_record_ros2_imd_cfg.yaml 使用，对于这个情况，需要将 `msg_type: sensor_msgs::msg::JointState` 改为 `msg_type: ros2:sensor_msgs/msg/JointState`

luojinghua50, 2025-05-14:
> 可以参考https://github.com/AimRT/AimRT/blob/main/src/examples/plugins/record_playback_plugin/install/linux/bin/cfg/examples_plugins_record_playback_plugin_record_ros2_imd_cfg.yaml 使用，对于这个情况，需要将 `msg_type: sensor_msgs::msg::JointState` 改为 `msg_type: ros2:sensor_msgs/msg/JointState`

感谢你的回复，验证成功了。不过再使用signal 模式时，遇到了一个新问题。我配置的
`      - name: record_playback_plugin
        path: ./libaimrt_record_playback_plugin.so
        options:
          timer_executor: storage_executor  # require time schedule!
          type_support_pkgs:       
            - path: ./libros2_msgs_ros2_ts.so
          record_actions:
            - name: my_signal_record
              options:
                bag_path: ./bag
                mode: signal # imd/signal
                max_preparation_duration_s: 10 # Effective only in signal mode
                executor: record_thread # require thread safe!
                ...
                topic_meta_list:
                  - topic_name: /zero_mode                   
                    msg_type: ros2:std_msgs/msg/Float32
                  - topic_name: /stand_mode
                    msg_type: ros2:std_msgs/msg/Float32
                  - topic_name: /walk_mode
                    msg_type: ros2:std_msgs/msg/Float32
                  - topic_name: /walk_mode2         
         `
对应的rpc配置
`  rpc: # rpc相关配置
    backends: # rpc后端配置
      - type: local
      - type: http
    clients_options: # 为 client 选择后端
      - func_name: "(.*)"
        enable_backends: [local]
    servers_options: # 为 server 选择后端
      - func_name: "(ros2:/aimrt.protocols.record_playback_plugin.*)"
        enable_backends: [http]`
执行http请求时候报错了
`data='{
    "action_name": "my_signal_record",
    "preparation_duration_s": 5,
    "record_duration_s": 10
}'

curl -i \
    -H 'content-type:application/json' \
    -X POST 'http://127.0.0.1:50080/rpc/aimrt.protocols.record_playback_plugin.RecordPlaybackService/StartRecord' \
    -d "$data"`
 
错误提示 HTTP/1.1 404 Not Found
Content-Type: text/html
Content-Length: 107
The resource '/rpc/aimrt.protocols.record_playback_plugin.RecordPlaybackService/StartRecord' was not found.     

查看aimrt 的src/protocols/plugins/record_playback_plugin/record_playback.proto 看到的这个定义，只支持pb方式吗 ？我需要做什么调整呢，期待得到你的答复。

luojinghua50, 2025-05-14:
> 可以参考https://github.com/AimRT/AimRT/blob/main/src/examples/plugins/record_playback_plugin/install/linux/bin/cfg/examples_plugins_record_playback_plugin_record_ros2_imd_cfg.yaml 使用，对于这个情况，需要将 `msg_type: sensor_msgs::msg::JointState` 改为 `msg_type: ros2:sensor_msgs/msg/JointState`

感谢你的回复，验证成功了。不过再使用signal 模式时，遇到了一个新问题。我配置的
     - name: record_playback_plugin
        path: ./libaimrt_record_playback_plugin.so
        options:
          timer_executor: storage_executor  # require time schedule!
          type_support_pkgs:       
            - path: ./libros2_msgs_ros2_ts.so
          record_actions:
            - name: my_signal_record
              options:
                bag_path: ./bag
                mode: signal # imd/signal
                max_preparation_duration_s: 10 # Effective only in signal mode
                executor: record_thread # require thread safe!
                ...
                topic_meta_list:
                  - topic_name: /zero_mode                   
                    msg_type: ros2:std_msgs/msg/Float32
                  - topic_name: /stand_mode
                    msg_type: ros2:std_msgs/msg/Float32  
对应的rpc配置
  rpc: # rpc相关配置
    backends: # rpc后端配置
      - type: local
      # - type: ros2
      - type: http
    clients_options: # 为 client 选择后端
      - func_name: "(.*)"
        enable_backends: [local]
    servers_options: # 为 server 选择后端
      - func_name: "(ros2:/aimrt.protocols.record_playback_plugin.*)"
        enable_backends: [http]

执行http请求时候报错了
`data='{
"action_name": "my_signal_record",
"preparation_duration_s": 5,
"record_duration_s": 10
}'

curl -i
-H 'content-type:application/json'
-X POST 'http://127.0.0.1:50080/rpc/aimrt.protocols.record_playback_plugin.RecordPlaybackService/StartRecord'
-d "$data"`

错误提示 HTTP/1.1 404 Not Found
Content-Type: text/html
Content-Length: 107
The resource '/rpc/aimrt.protocols.record_playback_plugin.RecordPlaybackService/StartRecord' was not found.

查看aimrt 的src/protocols/plugins/record_playback_plugin/record_playback.proto 看到的这个定义，只支持pb方式吗 ？我需要做什么调整呢，期待得到你的答复。

yglsaltfish, 2025-05-15:
这个是通过signal的方式来录包。具体表现为发送rpc请求，让录包插件开始录包，此处使用的是PB协议；
我没有复现你的问题，能提供完整的配置文件和执行http请求的文件吗

luojinghua50, 2025-05-15:
> 这个是通过signal的方式来录包。具体表现为发送rpc请求，让录包插件开始录包，此处使用的是PB协议； 我没有复现你的问题，能提供完整的配置文件和执行http请求的文件吗

我使用的是msg_type 和rpc 配置- func_name: "(ros2:/aimrt.protocols.record_playback_plugin.*)" 都是ros2前缀的，问题是不是出在这里不支持呢

这是X1的yaml 文件配置如下：
aimrt:
  plugin:
    plugins:
      - name: ros2_plugin
        path: ./libaimrt_ros2_plugin.so
        options:
          node_name: x1_rl_ctrl_node
          executor_type: MultiThreaded # SingleThreaded/StaticSingleThreaded/MultiThreaded
          executor_thread_num: 4
      - name: net_plugin
        path: ./libaimrt_net_plugin.so
        options:
          thread_num: 4
          http_options:
            listen_ip: 127.0.0.1
            listen_port: 50080

      - name: record_playback_plugin
        path: ./libaimrt_record_playback_plugin.so
        options:
          timer_executor: storage_executor  # require time schedule!
          type_support_pkgs:       
            - path: ./libros2_msgs_ros2_ts.so
          record_actions:
            - name: my_signal_record
              options:
                bag_path: ./bag
                mode: signal # imd/signal
                max_preparation_duration_s: 10 # Effective only in signal mode
                executor: record_thread # require thread safe!
                storage_policy:
                  storage_format: mcap
                  max_bag_size_m: 2048
                  msg_write_interval: 1000        # message count period
                  msg_write_interval_time: 1000   # ms
                  journal_mode: WAL
                  synchronous_mode: full
                  compression_mode: zstd     # comression mode
                  compression_level: default   # comression level
                ext_data:
                  - key: platform
                    value: arm64
                  - key: os
                    value: ubuntu
                topic_meta_list:
                  - topic_name: /idle_mode
                    msg_type: ros2:std_msgs/msg/Float32
                  - topic_name: /keep_mode
                    msg_type: ros2:std_msgs/msg/Float32 
                  - topic_name: /zero_mode                   
                    msg_type: ros2:std_msgs/msg/Float32
                  - topic_name: /stand_mode
                    msg_type: ros2:std_msgs/msg/Float32
                  - topic_name: /walk_mode
                    msg_type: ros2:std_msgs/msg/Float32
                  - topic_name: /walk_mode2
                    msg_type: ros2:std_msgs/msg/Float32 
                  - topic_name: /plan_mode                   
                    msg_type: ros2:std_msgs/msg/Float32 
                  - topic_name: /cmd_vel
                    msg_type: ros2:geometry_msgs/msg/Twist

          # playback_actions:
          #   - name: my_imd_playback
          #     options:
[... 83 lines cut ...]
          - JoyStickModule
          - ControlModule
          - SimModule
    modules: # 模块
      - name: JoyStickModule
        log_lvl: INFO
        cfg_file_path: ./cfg/joy_stick_module/joy_x1.yaml
      - name: ControlModule
        log_lvl: INFO
        cfg_file_path: ./cfg/control_module/rl_x1_sim.yaml
      - name: SimModule
        log_lvl: INFO
        cfg_file_path: ./cfg/sim_module/sim_x1.yaml

这是在终端执行的curl 命令：
data='{
    "action_name": "my_signal_record",
    "preparation_duration_s": 5,
    "record_duration_s": 10
}'

curl -i \
    -H 'content-type:application/json' \
    -X POST 'http://127.0.0.1:50080/rpc/aimrt.protocols.record_playback_plugin.RecordPlaybackService/StartRecord' \
    -d "$data"

yglsaltfish, 2025-05-15:
这部分应该使用pb前缀，这里使用的是protobuf协议
```yaml
      - func_name: "(pb:/aimrt.protocols.record_playback_plugin.*)"
```

PR #287 (2025-05-08): feat: add official type support pkg
close #281