AimRT/AimRT #291: [Bug] Brief what it is Ubuntu22.04 + Humble编译失败

XieWup, 2025-05-15:
**Version of AimRT**
eg. v1.0.0

**Environments**
versions of gcc, cmake, python etc.
gcc version 13.1.0
cmake version 3.28.1
Python 3.10.12

**Describe the bug**
A clear and concise description of what the bug is.
build.sh内容如下：
#!/bin/bash

# exit on error and print each command
set -e

if [ -d ./build/install ]; then
    rm -rf ./build/install
fi

cmake -B build \
    -DCMAKE_BUILD_TYPE=Release \
    -DAIMRT_INSTALL=ON \
    -DCMAKE_INSTALL_PREFIX=./build/install \
    -DAIMRT_BUILD_TESTS=OFF \
    -DAIMRT_BUILD_EXAMPLES=ON \
    -DAIMRT_BUILD_PROTOCOLS=ON \
    -DAIMRT_BUILD_DOCUMENT=OFF \
    -DAIMRT_BUILD_RUNTIME=ON \
    -DAIMRT_BUILD_CLI_TOOLS=OFF \
    -DAIMRT_BUILD_PYTHON_RUNTIME=OFF \
    -DAIMRT_USE_FMT_LIB=ON \
    -DAIMRT_BUILD_WITH_PROTOBUF=ON \
    -DAIMRT_USE_LOCAL_PROTOC_COMPILER=OFF \
    -DAIMRT_USE_PROTOC_PYTHON_PLUGIN=OFF \
    -DAIMRT_BUILD_WITH_ROS2=ON \
    -DAIMRT_BUILD_NET_PLUGIN=ON \
    -DAIMRT_BUILD_MQTT_PLUGIN=OFF \
    -DAIMRT_BUILD_ZENOH_PLUGIN=OFF \
    -DAIMRT_BUILD_ICEORYX_PLUGIN=OFF \
    -DAIMRT_BUILD_ROS2_PLUGIN=ON \
    -DAIMRT_BUILD_RECORD_PLAYBACK_PLUGIN=ON \
    -DAIMRT_BUILD_TIME_MANIPULATOR_PLUGIN=ON \
    -DAIMRT_BUILD_PARAMETER_PLUGIN=ON \
    -DAIMRT_BUILD_LOG_CONTROL_PLUGIN=ON \
    -DAIMRT_BUILD_TOPIC_LOGGER_PLUGIN=ON \
    -DAIMRT_BUILD_OPENTELEMETRY_PLUGIN=ON \
    -DAIMRT_BUILD_GRPC_PLUGIN=ON \
    -DAIMRT_BUILD_PYTHON_PACKAGE=OFF
    $@

cmake --build build --config Release --target install --parallel $(nproc)

终端输入：
source url_cn.bashrc
./build.sh $AIMRT_DOWNLOAD_FLAGS

终端输出如下：
/usr/bin/ld: ../../../libprotoc.a(plugin.pb.cc.o): in function `void google::protobuf::internal::RepeatedPtrFieldBase::MergeFromInnerLoop<google::protobuf::RepeatedPtrField<google::protobuf::FileDescriptorProto>::TypeHandler>(void**, void**, int, int)':
plugin.pb.cc:(.text._ZN6google8protobuf8internal20RepeatedPtrFieldBase18MergeFromInnerLoopINS0_16RepeatedPtrFieldINS0_19FileDescriptorProtoEE11TypeHandlerEEEvPPvS9_ii[_ZN6google8protobuf8internal20RepeatedPtrFieldBase18MergeFromInnerLoopINS0_16RepeatedPtrFieldINS0_19FileDescriptorProtoEE11TypeHandlerEEEvPPvS9_ii]+0x48): undefined reference to `google::protobuf::FileDescriptorProto* google::protobuf::Arena::CreateMaybeMessage<google::protobuf::FileDescriptorProto>(google::protobuf::Arena*)'
/usr/bin/ld: ../../../libprotoc.a(plugin.pb.cc.o): in function `google::protobuf::internal::GenericTypeHandler<google::protobuf::compiler::CodeGeneratorResponse_File>::Merge(google::protobuf::compiler::CodeGeneratorResponse_File const&, google::protobuf::compiler::CodeGeneratorResponse_File*)':
plugin.pb.cc:(.text._ZN6google8protobuf8internal18GenericTypeHandlerINS0_8compiler26CodeGeneratorResponse_FileEE5MergeERKS4_PS4_[_ZN6google8protobuf8internal18GenericTypeHandlerINS0_8compiler26CodeGeneratorResponse_FileEE5MergeERKS4_PS4_]+0x87): undefined reference to `google::protobuf::GeneratedCodeInfo::MergeImpl(google::protobuf::Message&, google::protobuf::Message const&)'
/usr/bin/ld: plugin.pb.cc:(.text._ZN6google8protobuf8internal18GenericTypeHandlerINS0_8compiler26CodeGeneratorResponse_FileEE5MergeERKS4_PS4_[_ZN6google8protobuf8internal18GenericTypeHandlerINS0_8compiler26CodeGeneratorResponse_FileEE5MergeERKS4_PS4_]+0x143): undefined reference to `google::protobuf::_GeneratedCodeInfo_default_instance_'
/usr/bin/ld: plugin.pb.cc:(.text._ZN6google8protobuf8internal18GenericTypeHandlerINS0_8compiler26CodeGeneratorResponse_FileEE5MergeERKS4_PS4_[_ZN6google8protobuf8internal18GenericTypeHandlerINS0_8compiler26CodeGeneratorResponse_FileEE5MergeERKS4_PS4_]+0x160): undefined reference to `google::protobuf::GeneratedCodeInfo* google::protobuf::Arena::CreateMaybeMessage<google::protobuf::GeneratedCodeInfo>(google::protobuf::Arena*)'
/usr/bin/ld: ../../../libprotoc.a(plugin.pb.cc.o):(.data.rel.ro+0x40): undefined reference to `descriptor_table_google_2fprotobuf_2fdescriptor_2eproto'
collect2: error: ld returned 1 exit status
gmake[2]: *** [src/tools/protoc_plugin_cpp_gen_aimrt_cpp_rpc/CMakeFiles/aimrt_tools_protoc_plugin_cpp_gen_aimrt_cpp_rpc.dir/build.make:100：protoc_plugin_cpp_gen_aimrt_cpp_rpc] 错误 1
gmake[1]: *** [CMakeFiles/Makefile2:6008：src/tools/protoc_plugin_cpp_gen_aimrt_cpp_rpc/CMakeFiles/aimrt_tools_protoc_plugin_cpp_gen_aimrt_cpp_rpc.dir/all] 错误 2
gmake[1]: *** 正在等待未完成的任务....
[ 56%] Linking CXX executable ../../../aimrt_main
[ 56%] Built target aimrt_runtime_main
gmake: *** [Makefile:166：all] 错误 2

XieWup, 2025-05-15:
快来人救救我！！

XieWup, 2025-05-15:
又编译了一次，使用命令：/build.sh $AIMRT_DOWNLOAD_FLAGS

终端报错：
collect2: error: ld returned 1 exit status
gmake[2]: *** [/home/liangs/AimRT/_deps/protobuf-build/CMakeFiles/protoc.dir/build.make:99：protoc-3.21.12.0] 错误 1
gmake[1]: *** [CMakeFiles/Makefile2:4671：/home/liangs/AimRT/_deps/protobuf-build/CMakeFiles/protoc.dir/all] 错误 2
[ 72%] Built target ros2_plugin_proto__py
[ 72%] Built target example_ros2__py
[ 72%] Built target aimrt_msgs__py
[ 72%] Linking CXX executable ../../../../../../native_ros2_rpc_server
[ 72%] Built target aimrt_examples_plugins_ros2_plugin_native_ros2_rpc_server
[ 72%] Linking CXX executable ../../../../../../native_ros2_rpc_client
[ 72%] Built target aimrt_examples_plugins_ros2_plugin_native_ros2_rpc_client
[ 72%] Linking CXX executable ../../../../../../native_ros2_chn_publisher
[ 72%] Built target aimrt_examples_plugins_ros2_plugin_native_ros2_chn_publisher
[ 72%] Linking CXX shared library ../../../../libros2_plugin_proto_ros2_ts.so
[ 72%] Built target aimrt_protocols_ros2_plugin_proto_ts
[ 72%] Linking CXX shared library ../../../../libexample_ros2_ros2_ts.so
[ 72%] Built target aimrt_protocols_example_ros2_ts
[ 72%] Linking CXX shared library ../../../../libaimrt_msgs_ros2_ts.so
[ 72%] Built target aimrt_protocols_aimrt_msgs_ts

[ 72%] Linking CXX static library ../../../../../../libaimrt_examples_cpp_ros2_chn_normal_subscriber_module.a
[ 72%] Built target aimrt_examples_cpp_ros2_chn_normal_subscriber_module
[ 72%] Linking CXX static library ../../../../libaimrt_protocols_example_ros2_aimrt_rpc_gencode.a
[ 72%] Built target aimrt_protocols_example_ros2_aimrt_rpc_gencode
[ 72%] Linking CXX executable ../../../../../../native_ros2_chn_subscriber
[ 72%] Built target aimrt_examples_plugins_ros2_plugin_native_ros2_chn_subscriber
[ 72%] Linking CXX static library ../../../../../../libaimrt_examples_cpp_ros2_chn_normal_publisher_module.a
[ 72%] Built target aimrt_examples_cpp_ros2_chn_normal_publisher_module
[ 72%] Linking CXX executable ../../../../../../normal_ros2_chn_subscriber_app
[ 72%] Built target aimrt_examples_cpp_ros2_chn_normal_ros2_chn_subscriber_app
[ 72%] Linking CXX static library ../../../../libaimrt_protocols_ros2_plugin_proto_aimrt_rpc_gencode.a
[ 72%] Built target aimrt_protocols_ros2_plugin_proto_aimrt_rpc_gencode
[ 72%] Linking CXX static library ../../../../../../libaimrt_examples_cpp_ros2_chn_benchmark_subscriber_module.a
[ 72%] Built target aimrt_examples_cpp_ros2_chn_benchmark_subscriber_module
[ 72%] Linking CXX static library ../../../../../../libaimrt_examples_cpp_ros2_chn_benchmark_publisher_module.a
[ 72%] Built target aimrt_examples_cpp_ros2_chn_benchmark_publisher_module
[ 72%] Linking CXX executable ../../../../../../normal_ros2_chn_publisher_app
[ 72%] Built target aimrt_examples_cpp_ros2_chn_normal_ros2_chn_publisher_app
[ 72%] Linking CXX shared library ../../../libaimrt_ros2_plugin.so
[ 72%] Built target aimrt_plugins_ros2_plugin
gmake: *** [Makefile:166：all] 错误 2

XieWup, 2025-05-16:
protoc --version : libprotoc 3.12.4

另外！我现在正在尝试二开灵犀X1，发现无法通过ros2 topic切换模式，是否通过AIMRT就可以通过topic切换X1的各种模式了？
安装并编译了https://github.com/AgibotTech/agibot_x1_infer 还需要在安装编译AimRT吗？

感谢回复！！！

zhangyi1357, 2025-05-22:
> protoc --version : libprotoc 3.12.4
> 
> 另外！我现在正在尝试二开灵犀X1，发现无法通过ros2 topic切换模式，是否通过AIMRT就可以通过topic切换X1的各种模式了？ 安装并编译了https://github.com/AgibotTech/agibot_x1_infer 还需要在安装编译AimRT吗？
> 
> 感谢回复！！！

据我所知，灵犀 x1 基于 AimRT 框架进行开发，使用 AimRT 肯定可以进行控制。不过 AimRT 本身兼容 ROS2，如果相应接口提供了 ROS2 后端应该是可以正常进行通信的，详细情况可以在 x1 仓库中提相应 issue。

zhangyi1357, 2025-05-22:
> **Version of AimRT** eg. v1.0.0
> 
> **Environments** versions of gcc, cmake, python etc. gcc version 13.1.0 cmake version 3.28.1 Python 3.10.12
> 
> **Describe the bug** A clear and concise description of what the bug is. build.sh内容如下：
> 
> #!/bin/bash #exit on error and print each command set -e
> 
> if [ -d ./build/install ]; then rm -rf ./build/install fi
> 
> cmake -B build -DCMAKE_BUILD_TYPE=Release -DAIMRT_INSTALL=ON -DCMAKE_INSTALL_PREFIX=./build/install -DAIMRT_BUILD_TESTS=OFF -DAIMRT_BUILD_EXAMPLES=ON -DAIMRT_BUILD_PROTOCOLS=ON -DAIMRT_BUILD_DOCUMENT=OFF -DAIMRT_BUILD_RUNTIME=ON -DAIMRT_BUILD_CLI_TOOLS=OFF -DAIMRT_BUILD_PYTHON_RUNTIME=OFF -DAIMRT_USE_FMT_LIB=ON -DAIMRT_BUILD_WITH_PROTOBUF=ON -DAIMRT_USE_LOCAL_PROTOC_COMPILER=OFF -DAIMRT_USE_PROTOC_PYTHON_PLUGIN=OFF -DAIMRT_BUILD_WITH_ROS2=ON -DAIMRT_BUILD_NET_PLUGIN=ON -DAIMRT_BUILD_MQTT_PLUGIN=OFF -DAIMRT_BUILD_ZENOH_PLUGIN=OFF -DAIMRT_BUILD_ICEORYX_PLUGIN=OFF -DAIMRT_BUILD_ROS2_PLUGIN=ON -DAIMRT_BUILD_RECORD_PLAYBACK_PLUGIN=ON -DAIMRT_BUILD_TIME_MANIPULATOR_PLUGIN=ON -DAIMRT_BUILD_PARAMETER_PLUGIN=ON -DAIMRT_BUILD_LOG_CONTROL_PLUGIN=ON -DAIMRT_BUILD_TOPIC_LOGGER_PLUGIN=ON -DAIMRT_BUILD_OPENTELEMETRY_PLUGIN=ON -DAIMRT_BUILD_GRPC_PLUGIN=ON -DAIMRT_BUILD_PYTHON_PACKAGE=OFF $@
> 
> cmake --build build --config Release --target install --parallel $(nproc)
> 
> 终端输入： ./build.sh
> 
> 终端输出如下： /usr/bin/ld: ../../../libprotoc.a(plugin.pb.cc.o): in function `void google::protobuf::internal::RepeatedPtrFieldBase::MergeFromInnerLoop<google::protobuf::RepeatedPtrField<google::protobuf::FileDescriptorProto>::TypeHandler>(void**, void**, int, int)': plugin.pb.cc:(.text._ZN6google8protobuf8internal20RepeatedPtrFieldBase18MergeFromInnerLoopINS0_16RepeatedPtrFieldINS0_19FileDescriptorProtoEE11TypeHandlerEEEvPPvS9_ii[_ZN6google8protobuf8internal20RepeatedPtrFieldBase18MergeFromInnerLoopINS0_16RepeatedPtrFieldINS0_19FileDescriptorProtoEE11TypeHandlerEEEvPPvS9_ii]+0x48): undefined reference to `google::protobuf::FileDescriptorProto* google::protobuf::Arena::CreateMaybeMessagegoogle::protobuf::FileDescriptorProto(google::protobuf::Arena*)' /usr/bin/ld: ../../../libprotoc.a(plugin.pb.cc.o): in function `google::protobuf::internal::GenericTypeHandler<google::protobuf::compiler::CodeGeneratorResponse_File>::Merge(google::protobuf::compiler::CodeGeneratorResponse_File const&, google::protobuf::compiler::CodeGeneratorResponse_File*)': plugin.pb.cc:(.text._ZN6google8protobuf8internal18GenericTypeHandlerINS0_8compiler26CodeGeneratorResponse_FileEE5MergeERKS4_PS4_[_ZN6google8protobuf8internal18GenericTypeHandlerINS0_8compiler26CodeGeneratorResponse_FileEE5MergeERKS4_PS4_]+0x87): undefined reference to `google::protobuf::GeneratedCodeInfo::MergeImpl(google::protobuf::Message&, google::protobuf::Message const&)' /usr/bin/ld: plugin.pb.cc:(.text._ZN6google8protobuf8internal18GenericTypeHandlerINS0_8compiler26CodeGeneratorResponse_FileEE5MergeERKS4_PS4_[_ZN6google8protobuf8internal18GenericTypeHandlerINS0_8compiler26CodeGeneratorResponse_FileEE5MergeERKS4_PS4_]+0x143): undefined reference to `google::protobuf::_GeneratedCodeInfo_default_instance_' /usr/bin/ld: plugin.pb.cc:(.text._ZN6google8protobuf8internal18GenericTypeHandlerINS0_8compiler26CodeGeneratorResponse_FileEE5MergeERKS4_PS4_[_ZN6google8protobuf8internal18GenericTypeHandlerINS0_8compiler26CodeGeneratorResponse_FileEE5MergeERKS4_PS4_]+0x160): undefined reference to `google::protobuf::GeneratedCodeInfo* google::protobuf::Arena::CreateMaybeMessagegoogle::protobuf::GeneratedCodeInfo(google::protobuf::Arena*)' /usr/bin/ld: ../../../libprotoc.a(plugin.pb.cc.o):(.data.rel.ro+0x40): undefined reference to `descriptor_table_google_2fprotobuf_2fdescriptor_2eproto' collect2: error: ld returned 1 exit status gmake[2]: *** [src/tools/protoc_plugin_cpp_gen_aimrt_cpp_rpc/CMakeFiles/aimrt_tools_protoc_plugin_cpp_gen_aimrt_cpp_rpc.dir/build.make:100：protoc_plugin_cpp_gen_aimrt_cpp_rpc] 错误 1 gmake[1]: *** [CMakeFiles/Makefile2:6008：src/tools/protoc_plugin_cpp_gen_aimrt_cpp_rpc/CMakeFiles/aimrt_tools_protoc_plugin_cpp_gen_aimrt_cpp_rpc.dir/all] 错误 2 gmake[1]: *** 正在等待未完成的任务.... [ 56%] Linking CXX executable ../../../aimrt_main [ 56%] Built target aimrt_runtime_main gmake: *** [Makefile:166：all] 错误 2

你提供的编译命令中似乎 -D 之间没有正确使用 \ 连接，可能导致某些选项未被正确传递进去？可以检查一下这些选项的值（在 CMake 中使用 message 输出对应的一些值，例如 AIMRT_BUILD_WITH_PROTOBUF 等关键变量）

zhangyi1357, 2025-06-05:
Close as no update for 2 weeks.